ckb-next/ckb-next #790: Constant 3% CPU usage

mysteryx93, 2022-02-10:
I'm glad to see (experimental) support for K95 Platinum XT, but a downside is that it takes a constant 3% CPU usage. Is this normal? Totally usable; but worth the cost?

For animations, I'd rather let the keyboard run in hardware mode to avoid the CPU usage; but being able to use the G keys would be great.

Is the 3% CPU usage normal, and any chance that this will improve in the future?

Because it's usable and probably "by design" because animations do require CPU work, I'm posting it as feature request instead of bug.

NRG-R9T, 2022-02-14:
You should provide your system specs and a log if you expect any help, if possible. It could be a config fault on your side by the OS or interference of programms, whatever.

I don't share your problem, here's with 0% CPU usage of ckb-next. I compiled it from source though and I don't have anything running with it on my K63 keyboard and my mouse (Dark Core RGB Pro SE) is still unsupported in Wireless mode (USB & Bluetooth run anyway, without ckb-next).

mysteryx93, 2022-02-14:
Good to know that the CPU usage isn't normal.

Here's my `inxi -Faz`
```
System:
Kernel: 5.16.8-zen1-1-zen x86_64 bits: 64 compiler: gcc v: 11.1.0
parameters: BOOT_IMAGE=/@/boot/vmlinuz-linux-zen
root=UUID=58203bda-e2e2-4c32-a006-c91d933cad4e rw rootflags=subvol=@
quiet splash rd.udev.log_priority=3 vt.global_cursor_default=0
systemd.unified_cgroup_hierarchy=1 loglevel=3 intel_iommu=on
mitigations=off
Desktop: KDE Plasma 5.24.0 tk: Qt 5.15.2 info: latte-dock wm: kwin_x11
vt: 1 dm: SDDM Distro: Garuda Linux base: Arch Linux
Machine:
Type: Laptop System: Acer product: Predator PH315-53 v: V1.01
serial: <superuser required>
Mobo: CML model: QX50_CMS v: V1.01 serial: <superuser required>
UEFI: Insyde v: 1.01 date: 04/27/2020
Battery:
ID-1: BAT1 charge: 41.8 Wh (100.0%) condition: 41.8/58.8 Wh (71.1%)
volts: 16.3 min: 15.4 model: SMP AP18E7M type: Li-ion serial: <filter>
status: Full
CPU:
Info: model: Intel Core i7-10750H bits: 64 type: MT MCP arch: Comet Lake
family: 6 model-id: 0xA5 (165) stepping: 2 microcode: 0xEC
Topology: cpus: 1x cores: 6 tpc: 2 threads: 12 smt: enabled cache:
L1: 384 KiB desc: d-6x32 KiB; i-6x32 KiB L2: 1.5 MiB desc: 6x256 KiB
L3: 12 MiB desc: 1x12 MiB
Speed (MHz): avg: 4645 high: 4802 min/max: 800/5000 scaling:
driver: intel_pstate governor: performance cores: 1: 4798 2: 4790 3: 4668
4: 3626 5: 4796 6: 4800 7: 4364 8: 4801 9: 4700 10: 4800 11: 4802
12: 4799 bogomips: 62399
Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Vulnerabilities:
Type: itlb_multihit status: KVM: VMX disabled
Type: l1tf status: Not affected
Type: mds status: Not affected
Type: meltdown status: Not affected
Type: spec_store_bypass status: Vulnerable
Type: spectre_v1 status: Vulnerable: __user pointer sanitization and
usercopy barriers only; no swapgs barriers
Type: spectre_v2 status: Vulnerable, IBPB: disabled, STIBP: disabled
Type: srbds status: Not affected
Type: tsx_async_abort status: Not affected
Graphics:
Device-1: Intel CometLake-H GT2 [UHD Graphics]
vendor: Acer Incorporated ALI driver: i915 v: kernel bus-ID: 00:02.0
chip-ID: 8086:9bc4 class-ID: 0300
Device-2: NVIDIA TU106M [GeForce RTX 2060 Mobile]
vendor: Acer Incorporated ALI driver: nvidia v: 510.47.03
alternate: nouveau,nvidia_drm bus-ID: 01:00.0 chip-ID: 10de:1f15
class-ID: 0300
Device-3: Quanta HD User Facing type: USB driver: uvcvideo bus-ID: 1-5:3
chip-ID: 0408:a061 class-ID: 0e02
Display: x11 server: X.Org 1.21.1.3 compositor: kwin_x11 driver:
loaded: modesetting,nvidia display-ID: :0 screens: 1
Screen-1: 0 s-res: 1920x1080 s-dpi: 96 s-size: 508x285mm (20.0x11.2")
s-diag: 582mm (22.9")
Monitor-1: HDMI-0 res: 1920x1080 hz: 60 dpi: 40
size: 1210x680mm (47.6x26.8") diag: 1388mm (54.6")
OpenGL: renderer: NVIDIA GeForce RTX 2060/PCIe/SSE2
v: 4.6.0 NVIDIA 510.47.03 direct render: Yes
Audio:
Device-1: Intel Comet Lake PCH cAVS vendor: Acer Incorporated ALI
driver: snd_hda_intel v: kernel
alternate: snd_soc_skl,snd_sof_pci_intel_cnl bus-ID: 00:1f.3
chip-ID: 8086:06c8 class-ID: 0403
Device-2: NVIDIA TU106 High Definition Audio
vendor: Acer Incorporated ALI driver: snd_hda_intel v: kernel
bus-ID: 01:00.1 chip-ID: 10de:10f9 class-ID: 0403
[... 39 lines cut ...]
Partition:
ID-1: / raw-size: 460.84 GiB size: 460.84 GiB (100.00%)
used: 271.47 GiB (58.9%) fs: btrfs dev: /dev/nvme0n1p5 maj-min: 259:3
ID-2: /boot/efi raw-size: 100 MiB size: 96 MiB (96.00%)
used: 51.2 MiB (53.4%) fs: vfat dev: /dev/nvme0n1p1 maj-min: 259:1
ID-3: /home raw-size: 460.84 GiB size: 460.84 GiB (100.00%)
used: 271.47 GiB (58.9%) fs: btrfs dev: /dev/nvme0n1p5 maj-min: 259:3
ID-4: /var/log raw-size: 460.84 GiB size: 460.84 GiB (100.00%)
used: 271.47 GiB (58.9%) fs: btrfs dev: /dev/nvme0n1p5 maj-min: 259:3
ID-5: /var/tmp raw-size: 460.84 GiB size: 460.84 GiB (100.00%)
used: 271.47 GiB (58.9%) fs: btrfs dev: /dev/nvme0n1p5 maj-min: 259:3
Swap:
Kernel: swappiness: 133 (default 60) cache-pressure: 100 (default)
ID-1: swap-1 type: zram size: 15.46 GiB used: 562.2 MiB (3.6%)
priority: 100 dev: /dev/zram0
Sensors:
System Temperatures: cpu: 61.0 C pch: 69.0 C mobo: N/A gpu: nvidia
temp: 57 C
Fan Speeds (RPM): N/A
Info:
Processes: 363 Uptime: 1d 11h 54m wakeups: 1 Memory: 15.46 GiB
used: 3.73 GiB (24.1%) Init: systemd v: 250 tool: systemctl Compilers:
gcc: 11.1.0 clang: 13.0.1 Packages: pacman: 1792 lib: 496 Shell: fish
v: 3.3.1 default: Bash v: 5.1.16 running-in: konsole inxi: 3.3.12
```

Ravenslofty, 2022-04-26:
So, there are three things here.

1. Yes, using some small amount of CPU time doing software animations is normal. This is... At least partially because of legacy design decisions of Corsair and then ckb-next: it was only fairly recently (the original K95 Platinum and Strafe RGB) that Corsair devices could do hardware animation at all, so ckb-next (or CUE) had to do all the rendering in software.
2. But yes, the Platinum and Platinum XT can do hardware animations; unfortunately I got about halfway through implementing that for the original Platinum and found out that it's an absolute nightmare (CUE uploads like a megabyte of DEFLATEd XML and I can't tell if that's required or not, plus possible binary blobs?) that makes implementing...risky.
3. And the XT changes from the NXP protocol to the Bragi protocol, so it's almost certainly something completely different to my efforts to add support for the original Platinum.

I know it's not great to hear that even though your keyboard can do hardware animations, it isn't doing them (especially since you're on a laptop). Perhaps eventually we'll have support for them. But software rendering does have some small advantage: with things like the `pipe` animation, you can get your keyboard to notify you when things happen.

mysteryx93, 2022-05-21:
As a note, although `htop` reports 3% CPU usage, `System Monitor` reports less than 0.1% CPU usage. Thus it seems it's fine.

As for hardware mode, I can set it up using a Windows VM.

I think we can close this then.

closed 2022-05-21 by mysteryx93