ckb-next/ckb-next #791: HELP!!! Daemon not working

pinkturtle777, 2022-02-12:
Hello, I recently bought a Consair Strafe Cherry MX with Blue Switches off a friend, and quickly realized that the ICue program was not available for ChromeOS or Linux. I spent hours yesterday trying to get ckb-next to work. I am at the point where I can type "ckb-next" in the terminal and it will open, but I get a message saying the daemon is not working, which means that the program still doesn't work. I type in "sudo systemctl start ckb-next-daemon" but it doesn't do anything. I'm really new to Linux in general, so any help is appreciated.

pinkturtle777, 2022-02-12:
[image]

alvesvaren, 2022-02-12:
Run `sudo systemctl status ckb-next-daemon` and see if it is running or why it crashed

pinkturtle777, 2022-02-12:
@alvesvaren 
[image]

GhostMonkeyFPS, 2022-04-22:
I'm not an expert but. It appears that ckb cannot create /dev/input/ckb0 directory. Does its parent directory exists?

Ravenslofty, 2022-04-26:
`/dev/input` should exist for other things. Perhaps it's more that ChromeOS has some rule forbidding creation of directories in there?

I know that OpenSUSE carries a patch that changes the directory to `/var/run` by changing [this](https://github.com/ckb-next/ckb-next/blob/master/src/daemon/devnode.c#L13) and [this](https://github.com/ckb-next/ckb-next/blob/5a279b7bb2e97c7dceece12bd7e164d3372a4a81/src/gui/kbmanager.cpp#L6). It's... maybe worth a shot?