ckb-next/ckb-next #792: Force device redetection of devices so a system restart is not required would make this software perfect in my eyes

BAZOOKAPANDA-REAL, 2022-02-12:
**## What is the program's current behavior?**
I run Ubuntu as a VM on Unraid along side windows as such when I disconnect my keyboard k95 18 key to use for another VM and 
than re connect it ckb will not see my keyboard
(application restart will not do the trick unfortunately (even through terminal), a system restart is necessary)

**## What would you like it to do instead?**
Having a button to force re-detect devices would be wonderful if possible
maybe being able to set a time interval for re-detection

**## Any other notes & comments?**
If it is at all possible to automate the re-detection process would be the ideal but if not maybe being able to set a time interval for when the software checks for devices could bridge the gab between the convenience of auto re-detection (which would be difficult I'm sure) and manually re-detecting.

That is about it.
**Also this software is a god send so thank you.**

closed 2022-02-17 by BAZOOKAPANDA-REAL

Ravenslofty, 2022-02-18:
If that doesn't work, make a report while ckb-next isn't detecting the keyboard - there's a button for it.

BAZOOKAPANDA-REAL, 2022-02-23:
> daemon

What is the command to restart the daemon?

Ravenslofty, 2022-02-23:
`sudo systemctl restart ckb-next-daemon`

BAZOOKAPANDA-REAL, 2022-02-23:
> `sudo systemctl restart ckb-next-daemon`

Thanks.

I tried this and is did cause the devices to be re-detected.
Would like to not do this each time but its miles better than a restart so i have made a not and will issue this command whenever I detach and reattach my devices.

Thank you.

Ravenslofty, 2022-02-23:
Okay, so, that narrows the issue down a bit.

There's another command which can monitor the output of the daemon: `systemctl status ckb-next-daemon`. If possible, could you detach your keyboard, and then copy/paste the output of that command here?

BAZOOKAPANDA-REAL, 2022-02-23:
So first I detached and reattached they keyboard through unraid ans this is running as a VM. the device was re-detected so if that behavior continues than the issue would be solved

Secondly I physically unplugged the device and passed the device back through and the device was than not re-detected.
I ran the command and received:
chris@chris-Ubuntu:~$ ckb-next-daemon[19614]: [I] Stopping indicator thread for ckb2 (4)
bash: syntax error near unexpected token `('
chris@chris-Ubuntu:~$ Feb 23 13:44:06 chris-Ubuntu ckb-next-daemon[19614]: [I] Disconnecting /dev/input/ckb2
Feb: command not found
chris@chris-Ubuntu:~$ Feb 23 13:44:06 chris-Ubuntu ckb-next-daemon[19614]: [W] usbunclaim (usb_linux.c:459): Your Corsair K95 RGB Gaming Keyboard is being uncleanly removed to sp>
bash: syntax error near unexpected token `('
chris@chris-Ubuntu:~$ Feb 23 13:44:06 chris-Ubuntu ckb-next-daemon[19614]: [W] usbunclaim (usb_linux.c:460): If you still need the device, you will have to restart ckb-next-daemo>
bash: syntax error near unexpected token `('
chris@chris-Ubuntu:~$ Feb 23 13:44:06 chris-Ubuntu ckb-next-daemon[19614]: [I] Removed device path /dev/input/ckb2
Feb: command not found
chris@chris-Ubuntu:~$ Feb 23 13:44:06 chris-Ubuntu ckb-next-daemon[19614]: [I] Removed device path /dev/input/ckb2
Feb: command not found
chris@chris-Ubuntu:~$ Feb 23 13:44:08 chris-Ubuntu ckb-next-daemon[19614]: [I] Connecting Corsair K95 RGB Gaming Keyboard at /dev/input/ckb2
Feb: command not found
chris@chris-Ubuntu:~$ Feb 23 13:44:08 chris-Ubuntu ckb-next-daemon[19614]: [I] Starting input thread for /dev/input/ckb2
Feb: command not found
chris@chris-Ubuntu:~$ Feb 23 13:44:08 chris-Ubuntu ckb-next-daemon[19614]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 08 03 11 00 1c 1b 11 1b 04 01 03 06 c0 ff 4>
bash: syntax error near unexpected token `newline'
chris@chris-Ubuntu:~$ Feb 23 13:44:09 chris-Ubuntu ckb-next-daemon[19614]: [I] Setup finished for /dev/input/ckb2
Feb: command not found
chris@chris-Ubuntu:~$ 

This is different to what I received when software detaching the device and reattaching. 
What I will do next is restart the deamon see if it is connected than software detach the device pass is through to my win 10 vm and than detach and reattach and see what happens.

Ravenslofty, 2022-02-23:
I didn't mean paste it into your terminal, I meant paste it into GitHub so I can read it.

BAZOOKAPANDA-REAL, 2022-02-23:
Past the results of this command "systemctl status ckb-next-daemon"?

Ravenslofty, 2022-02-23:
Yes

BAZOOKAPANDA-REAL, 2022-02-23:
If there is a specific method of copying and a specific place you want me to past the output that I am not aware of pls tell me (I'm a bit of a novice in that way) 
Thanks

What I posted above was the first instance of the output I see (In the case where the device was physically unplugged, plugged back in and passed through)

The second instance of what I see (in the case where the device is software disconnected and passed through to win 10 vm and passed back into ubuntu) is as follows:

chris@chris-Ubuntu:~$ systemctl status ckb-next-daemon
● ckb-next-daemon.service - Corsair Keyboards and Mice Daemon
     Loaded: loaded (/lib/systemd/system/ckb-next-daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-02-23 13:53:42 SAST; 6min ago
   Main PID: 22860 (ckb-next-daemon)
      Tasks: 7 (limit: 4627)
     Memory: 1.2M
     CGroup: /system.slice/ckb-next-daemon.service
             └─22860 /usr/bin/ckb-next-daemon

Ubuntu ckb-next-daemon[22860]: [I] Disconnecting /dev/input/ckb2

Ubuntu ckb-next-daemon[22860]: [I] Caught internal signal SIGUSR2

Ubuntu ckb-next-daemon[22860]: [I] Stopping indicator thread for ckb2 (4)

Ubuntu ckb-next-daemon[22860]: [W] usbunclaim (usb_linux.c:459): Your Corsair K95 RGB Gaming Keyboard is being 
uncleanly removed to speed up shutdown times.

Ubuntu ckb-next-daemon[22860]: [W] usbunclaim (usb_linux.c:460): If you still need the device, you will have to restart ckb-next-daemon.

Ubuntu ckb-next-daemon[22860]: [I] Removed device path /dev/input/ckb2

Ubuntu ckb-next-daemon[22860]: [I] Connecting Corsair K95 RGB Gaming Keyboard at /dev/input/ckb2

Ubuntu ckb-next-daemon[22860]: [I] Starting input thread for /dev/input/ckb2

Ubuntu ckb-next-daemon[22860]: [I] Received identification packet: 0e 01 00 00 01 01 00 01 08 03 11 00 1c 1b 11 1b 04 01 03 06 c0 ff 40 00 03 00 00 00 02 44 00 00 00 00 00 >

Ubuntu ckb-next-daemon[22860]: [I] Setup finished for /dev/input/ckb2
lines 1-19/19 (END)

Ravenslofty, 2022-02-23:
This is absolutely fine, thank you.

Hmm, the unclean removal is a good hint as to what the problem might be, however I would probably also need this when the keyboard does not reattach, which you mentioned was fixed when the daemon was restarted.

If it happens again, please run that `systemctl status ckb-next-daemon` command again and paste the output. My suspicion here is that the unclean removal warning will be there as well, but let's wait and see.

BAZOOKAPANDA-REAL, 2022-02-23:
Will do.

Should mention that the first bit of code that I first pasted was when I physically unplugged they keyboard and plugged it back in and in that the the motherboard was net re-detected when I passed it through.

But I do also have a corsair mouse so I was expecting to see similar output to my second output post which is weird as I went into the gui and saw the mouse listed there but no keyboard.

Strange.

Non the less swapping between VM's seems to be flawless without needing to restart the deamon every time so in a way my problem is solved so thanks.

BAZOOKAPANDA-REAL, 2022-02-25:
Ran this (systemctl status ckb-next-daemon) just before restarting the deamon when they keyboard was connected and passed through but not detected by ckb.

Ubuntu:~$ systemctl status ckb-next-daemon
● ckb-next-daemon.service - Corsair Keyboards and Mice Daemon
     Loaded: loaded (/lib/systemd/system/ckb-next-daemon.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2022-02-24 19:33:32 SAST; 21h ago
   Main PID: 7458 (ckb-next-daemon)
      Tasks: 3 (limit: 4627)
     Memory: 980.0K
     CGroup: /system.slice/ckb-next-daemon.service
             └─7458 /usr/bin/ckb-next-daemon

Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [W] usbunclaim (usb_linux.c:460): If you still need the device, you will have>
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [E] os_resetusb (via usb.c:614): resetusb failed: No such device
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [E] usb_tryreset (usb.c:622): Reset failed. Disconnecting.
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [I] Disconnecting /dev/input/ckb1
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [I] Caught internal signal SIGUSR2
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [I] Stopping indicator thread for ckb1 (4)
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [I] Disconnecting /dev/input/ckb1
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [W] usbunclaim (usb_linux.c:459): Your Corsair K95 RGB Gaming Keyboard is bei>
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [W] usbunclaim (usb_linux.c:460): If you still need the device, you will have>
Feb 25 11:40:51 Ubuntu ckb-next-daemon[7458]: [I] Removed device path /dev/input/ckb1

Ravenslofty, 2022-02-25:
That log confirms my suspicion, but could you run something else for me, please?

`ckb-next-daemon --version`

BAZOOKAPANDA-REAL, 2022-02-25:
ckb-next: Corsair RGB driver 0.4.4

Ravenslofty, 2022-02-25:
It's the latest release, but that release was made in March of last year.

Hmmm, I need to think about this.

Ravenslofty, 2022-02-25:
Here's my thought process:

ckb-next-daemon gets notified about new USB devices being plugged in from udev. If udev notifies ckb-next-daemon, you'd see a corresponding "[I] Connecting ..." line. Of course, ckb-next can't manage your keyboard if it doesn't know about it.

What makes this strange is that restarting the daemon picks up the device, which implies that udev knows about the device so that it can tell the daemon.

Restart the daemon, then in a separate terminal, can you have `udevadm monitor` running, and then try removing and reattaching the keyboard until the daemon breaks, and paste the output of the `udevadm monitor` command into GitHub.

BAZOOKAPANDA-REAL, 2022-02-25:
Restarted and than reattached until the lighting profile I set up in ckb was no longer loading.
(should mention that a did one last detach and reattach to make sure it wasn't a fluke)
Finally ctrl c to exit
Output as follows

Ubuntu:~$ udevadm monitor
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[98191.990144] unbind   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
KERNEL[98191.992621] remove   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
KERNEL[98191.992734] unbind   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.1 (usb)
KERNEL[98191.992783] remove   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.1 (usb)
KERNEL[98191.992846] unbind   /devices/pci0000:00/0000:00:07.0/usb1/1-3 (usb)
KERNEL[98191.992901] remove   /devices/pci0000:00/0000:00:07.0/usb1/1-3 (usb)
UDEV  [98191.997840] unbind   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.1 (usb)
UDEV  [98191.998540] unbind   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
UDEV  [98191.999872] remove   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.1 (usb)
UDEV  [98192.000952] remove   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
UDEV  [98192.004331] unbind   /devices/pci0000:00/0000:00:07.0/usb1/1-3 (usb)
UDEV  [98192.006236] remove   /devices/pci0000:00/0000:00:07.0/usb1/1-3 (usb)
KERNEL[98192.151934] remove   /devices/virtual/input/input164/event14 (input)
UDEV  [98192.155095] remove   /devices/virtual/input/input164/event14 (input)
KERNEL[98192.171705] change   /devices/virtual/input/input164/input164::numlock (leds)
KERNEL[98192.171820] remove   /devices/virtual/input/input164/input164::numlock (leds)
KERNEL[98192.171905] change   /devices/virtual/input/input164/input164::capslock (leds)
KERNEL[98192.171993] remove   /devices/virtual/input/input164/input164::capslock (leds)
KERNEL[98192.172078] change   /devices/virtual/input/input164/input164::scrolllock (leds)
KERNEL[98192.172216] remove   /devices/virtual/input/input164/input164::scrolllock (leds)
UDEV  [98192.172909] change   /devices/virtual/input/input164/input164::numlock (leds)
UDEV  [98192.173028] change   /devices/virtual/input/input164/input164::capslock (leds)
UDEV  [98192.174490] remove   /devices/virtual/input/input164/input164::capslock (leds)
UDEV  [98192.175225] remove   /devices/virtual/input/input164/input164::numlock (leds)
UDEV  [98192.178080] change   /devices/virtual/input/input164/input164::scrolllock (leds)
UDEV  [98192.179454] remove   /devices/virtual/input/input164/input164::scrolllock (leds)
KERNEL[98192.187389] remove   /devices/virtual/input/input164 (input)
KERNEL[98192.187488] remove   /devices/virtual/input/input165/mouse2 (input)
UDEV  [98192.191190] remove   /devices/virtual/input/input164 (input)
UDEV  [98192.191327] remove   /devices/virtual/input/input165/mouse2 (input)
KERNEL[98192.222918] remove   /devices/virtual/input/input165/event15 (input)
UDEV  [98192.226944] remove   /devices/virtual/input/input165/event15 (input)
KERNEL[98192.248241] remove   /devices/virtual/input/input165 (input)
UDEV  [98192.251216] remove   /devices/virtual/input/input165 (input)
KERNEL[98195.117319] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3 (usb)
KERNEL[98195.118577] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
KERNEL[98195.125671] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035 (hid)
KERNEL[98195.126480] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168 (input)
KERNEL[98195.126863] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/event14 (input)
KERNEL[98195.126957] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/input168::numlock (leds)
KERNEL[98195.127032] change   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/input168::numlock (leds)
KERNEL[98195.127203] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/input168::capslock (leds)
KERNEL[98195.127601] change   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/input168::capslock (leds)
KERNEL[98195.127684] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/input168::scrolllock (leds)
KERNEL[98195.127767] change   /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input168/input168::scrolllock (leds)
KERNEL[98195.127859] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input169 (input)
KERNEL[98195.127934] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input169/event15 (input)
KERNEL[98195.128042] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input170 (input)
KERNEL[98195.128162] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/input/input170/event18 (input)
KERNEL[98195.128393] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/usbmisc/hiddev1 (usbmisc)
KERNEL[98195.128546] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035/hidraw/hidraw2 (hidraw)
KERNEL[98195.128641] bind     /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.0035 (hid)
KERNEL[98195.128753] bind     /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
[... cut ...]
UDEV  [98231.381419] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.003D/input/input186/event14 (input)
UDEV  [98231.421030] add      /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.003D/input/input187/event15 (input)
UDEV  [98231.424147] bind     /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0/0003:1B1C:1B11.003D (hid)
UDEV  [98231.426259] bind     /devices/pci0000:00/0000:00:07.0/usb1/1-3/1-3:1.0 (usb)
UDEV  [98231.433693] bind     /devices/pci0000:00/0000:00:07.0/usb1/1-3 (usb)
^C