ckb-next/ckb-next #793: Failed to start ckb-next-daemon.service: Unit ckb-next-daemon.service is masked.

guzi-momma, 2022-02-17:
i have used the recommend solution that pops up when opening the application and i can not get this to work

alvesvaren, 2022-02-23:
Try unmasking it with `sudo systemctl unmask ckb-next-daemon`

MaoMaoCake, 2022-06-06:
Can we have this added to the wiki as part of the installation step? This should help newbies to actually get it to work.

tatokis, 2022-06-12:
I don't know why this is happening, however if you're seeing this with the packages from the ubuntu/debian repositories (instead of the PPA), you should open a bug there instead. We do not provide support for these packages.

That said, I'll add the unmask command in the instructions and in the packaging for the PPA. Running it when it's not needed won't harm.

commit 5e5ce746 (2022-06-12): Add warning about daemon being masked and unmask it

Closes #793
closed 2022-06-12 by tatokis