ckb-next/ckb-next #799: Outdated wiki

fracasula, 2022-03-03:
In [this page](https://github.com/ckb-next/ckb-next/wiki/Linux-Installation) for OpenSuse there is a missing packet.

What you are reporting is to do the following:
```
sudo zypper install gcc gcc-c++ make cmake linux-glibc-devel zlib-devel libqt5-qtbase-devel systemd-devel libpulse-devel quazip-qt5-devel libudev-devel libqt5-linguist-devel git libdbusmenu-qt5-devel

git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
./quickinstall
```

But then you would get:
```
CMake Error at src/gui/CMakeLists.txt:287 (message):
  XCB-EWMH was not found.  Either install it or pass -DUSE_XCB=0 to disable
  active window detection.
```

You're missing this packet: `sudo zypper in xcb-util-wm-devel`.

I'm running on the latest OpenSuse Tumbleweed (rolling release).

tatokis, 2022-03-03:
Thank you.

It's probably also missing libxcb-devel and xcb-util-wm-devel. Can you verify that those packages exist in the repositories to add them as well?

tatokis, 2022-03-03:
Thank you.

It's probably also missing libxcb-devel and xcb-util-devel. Can you verify that those packages exist in the repositories to add them as well?

fracasula, 2022-03-03:
I had `libxcb-devel` already installed and the package name is correct if you want to add it.
In regards to `xcb-util-devel`, also the package name is correct and exists in the repos but I didn't have to install it (and the drivers are working).

Thanks for all this by the way, I can finally configure my K95 on Linux :slightly_smiling_face:

tatokis, 2022-03-03:
Awesome! Just edited the page in the wiki.

closed 2022-03-03 by tatokis

fracasula, 2022-03-19:
I ended up building on a new system and I got this error:

```
--   Found xcb-screensaver, version 1.14
CMake Error at src/gui/CMakeLists.txt:275 (message):
  XCB-Screensaver or Qt5 X11Extras was not found.  Either install them or
  pass -DUSE_XCB=0 to disable idle detection.
```

Solved by installing `libqt5-qtx11extras-devel` :+1:

tatokis, 2022-03-30:
Thanks. Added that as well