ckb-next/ckb-next #801: Corsair Scimitar RGB ELITE missing functionality

BadLuq, 2022-03-20:
Cannot map 1, 4, 7, or 10. The other 8 buttons work as intended.

tatokis, 2022-03-29:
You'll need to build from source for this.
First, install the dependencies from wiki (https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source), then:
```
git clone https://github.com/ckb-next/ckb-next
cd ckb-next
mkdir build
cd build
cmake .. -DDEBUG_USB_INPUT=1
make -j4 daemon
sudo systemctl stop ckb-next-daemon
sudo ./bin/ckb-next-daemon
```

Before you run the last command, make sure the GUI is running and hold your mouse in the air (so that the cursor doesn't move and spam messages).
Once the daemon is running in the terminal, press all of the buttons one by one slowly, starting from 1.

Once you finish that, reply here with the output of the daemon in the terminal.
You can then Ctrl C the daemon and restart the system one with `sudo systemctl start ckb-next-daemon`.

BadLuq, 2022-03-29:
<pre>[j@ICE build]$ sudo ./bin/ckb-next-daemon
ckb-next: Corsair RGB driver 0.4.4-190-geb8a1d1
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR SCIMITAR RGB ELITE Gaming Mouse at /dev/input/ckb1
[I] Connecting CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard at /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/ep_81
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ep_81
[I] ckb1 Input: (81) 0e 01 00 00 01 01 00 01 20 03 06 03 1c 1b 8b 1b 01 01 01 30 c1 ff 40 40 01 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 00 01 55 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb2 Input: (81) 0e 01 00 00 01 01 00 01 24 03 03 03 1c 1b 49 1b 01 01 00 01 c0 ff 40 00 03 00 00 00 02 40 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Received identification packet: 0e 01 00 00 01 01 00 01 24 03 03 03 1c 1b 49 1b 01 01 00 01 c0 ff 40 00 03 00 00 00 02 40 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Received identification packet: 0e 01 00 00 01 01 00 01 20 03 06 03 1c 1b 8b 1b 01 01 01 30 c1 ff 40 40 01 00 01 01 00 00 00 00 00 00 00 00 01 00 00 01 00 00 02 01 00 01 55 00 6c dc 02 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Setup finished for /dev/input/ckb2
[I] Setup finished for /dev/input/ckb1
[I] ckb1 Input: (81) 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[E] process_input_urb (keymap.c:674): Unknown mouse data received in input thread 10 from endpoint 81
[I] ckb1 Input: (81) 02 00 00 00 00 
[E] process_input_urb (keymap.c:674): Unknown mouse data received in input thread 02 from endpoint 81
[I] ckb1 Input: (81) 01 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
[I] ckb2 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I] Stopping input thread for /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Disconnecting /dev/input/ckb2
[I] Caught internal signal SIGUSR2
[I] Stopping indicator thread for ckb2 (4)
[I] Stopping input thread for /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
[I] Closing root controller
[I] Removed device path /dev/input/ckb0
</pre>

BadLuq, 2022-03-29:
Just wanna add that every time I pressed a button it did output something to the terminal, so could it just be a gui issue?

tatokis, 2022-04-01:
[image]
Is this how it looks for you?
The button column on the left is completely cut off...

BadLuq, 2022-04-01:
yes thats exactly how it looks for me. Looking back I should have clarified that in my post, sorry about that.

tatokis, 2022-04-01:
Can you please test the latest code from master? It should be fixed by https://github.com/ckb-next/ckb-next/commit/539dc64a25cea4ef89225b62bc9501bb0822f083.

BadLuq, 2022-04-01:
Fixed! thank you very much

closed 2022-04-01 by BadLuq