ckb-next/ckb-next #803: Assigning CAPS LOCK to ESC and the FN to CAPS LOCK. FN then outputs ESC when pressed

rehaby, 2022-03-23:
Using a STRAFE keyboard and the package ckb-next version 0.4.4-1~focal

* I set CAPS LOCK to output ESC and that works
* I then set FN to output CAPS LOCK
* After that, when I press FN, I get ESC

I also tried setting CAPS LOCK + FN to be CAPS LOCK and that unset CAPS LOCK from being ESC

tatokis, 2022-04-04:
I can't reproduce this. Can you try with [ckb-next-git](https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git)?

awvalenti, 2024-11-21:
Just curious, does Corsair originally allow remmaping Fn key in hardware? I own a Strafe RGB MK2 without Fn keys and always wondered whether that would be possible, cause I'd miss the context menu key, which was replaced by Fn on newer models. Thanks!