ckb-next/ckb-next #804: Installation failed

Maxiferado, 2022-03-25:
Fist hand i would like to thanks for you program. i used to use [ccMSC].
i followed every single step on the installation, and i can-t install the program i saw the last commit and i could see something related to it i'm not and expert so i must be wrong. 
i hope you can give me a sign of what can i do to fixed.

*Ubuntu 20.04.4 LTS*
[bug.txt](https://github.com/ckb-next/ckb-next/files/8354289/bug.txt)

Knirps1, 2022-03-26:
I have the same error on up to date Arch using the ckb-next-git package from the aur.

SoyCamel, 2022-03-26:
I'm having the same issue with Debian 11

tatokis, 2022-03-26:
Fixed by https://github.com/ckb-next/ckb-next/commit/eb8a1d13f98b42d62d4698e738408c0f361cdabc

closed 2022-03-26 by tatokis