ckb-next/ckb-next #818: Error Loading ckb

thesnowfury, 2022-04-17:
Hi getting this after installing ckb

ckb-next: error while loading shared libraries: libquazip1-qt5.so.1.0.0: cannot open shared object file: No such file or directory

any ideas?

thesnowfury, 2022-04-17:
Nevermind, QuaZip got updated, downgraded to 1.0 and all working again

closed 2022-04-17 by thesnowfury

tatokis, 2022-04-18:
What program did you use to install ckb-next? pamac? yay?

zeparu, 2022-04-18:
dont even need to downgrade, i just symlinked it to libquazip1-qt5.so.1.0.0, still works fine

issue shouldn't be closed imo, maintainers should upgrade / change the dependency with libquazip

thesnowfury, 2022-04-19:
> What program did you use to install ckb-next? pamac? yay?

I used pacman to install a while ago but it broke when I did a system update (sudo pacman -Syu)

vrpsunited, 2022-04-19:
> dont even need to downgrade, i just symlinked it to libquazip1-qt5.so.1.0.0, still works fine
> 
> issue shouldn't be closed imo, maintainers should upgrade / change the dependency with libquazip

Good idea - did you simply create a file with the name of the old lib and inside the path to the new lib?
:D