ckb-next/ckb-next #820: Update broke k95 RGB Platinum XT Support

rrediske, 2022-04-22:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**
```see attached```

## General information

### Source
How and where did you get this program, e.g.:
  * Package from AUR/PPA/COPR/...
  * Built manually using quickinstall script
  * Pre-built dmg for macOS

```Pamac, Manjaro Package Manager ```

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?

```
The ckb-next GUI no longer shows my Corsair K95 RGB Platinum XT keyboard.
I added the  "--enable-experimental" flag back into the systemd unit file manually
I noticed the binaries have also moved:

[onyx ~]# ls -l /usr/local/bin/ckb-next*
-rwxr-xr-x 1 root root 2754864 Mar 26 10:48 /usr/local/bin/ckb-next
-rwxr-xr-x 1 root root  225760 Mar 26 10:48 /usr/local/bin/ckb-next-daemon
-rwxr-xr-x 1 root root    3434 Mar 26 10:48 /usr/local/bin/ckb-next-dev-detect
[onyx ~]# ls -l /usr/bin/ckb-next*
-rwxr-xr-x 1 root root 2275440 Apr 22 17:16 /usr/bin/ckb-next
-rwxr-xr-x 1 root root  154240 Apr 22 17:16 /usr/bin/ckb-next-daemon
-rwxr-xr-x 1 root root    3404 Apr 22 17:16 /usr/bin/ckb-next-dev-detect
```

## What is the expected behaviour of the program?
```
All the features I used were working as expected before the upgrade, except for the firmware update.
I was able to assign keys, change colors, etc.
```

## Any other notes & comments?

[ckb-next-report-1650666314.gz](https://github.com/ckb-next/ckb-next/files/8544400/ckb-next-report-1650666314.gz)

rrediske, 2022-04-22:
I resolved my issue by removing the ckb-next package with pamac and installing the ckb-next-git package again. I had to enable the ckb-next-daemon service again, manually add the --enable-experimental flag to the systemd unit file again and I still have two different versions between /usr/bin and /usr/local/bin - but it works right now as far as I can tell so far.

Ravenslofty, 2022-04-26:
The `/usr/local` binaries suggest you probably built ckb-next from source without the package manager. We provide an uninstall build target for that.

As for it breaking: you probably hit the issue where QuaZip broke the library ABI, causing a soname change which then means that the dynamic linker can't find the old library anymore. Rebuilding ckb-next is the fix.

closed 2022-04-26 by Ravenslofty