ckb-next/ckb-next #823: Change the increments values for the options in settings...

Luxcium, 2022-05-03:
## What is the program's current behaviour?
In a section where a parameter is a *double* when using arrows from *GUI*  to set a value the minimum value goes directly to 0.1 and then up to 1.1 
## What would you like it to do instead?
It would be appreciated (if it is easy to implement) to assess if the value is between 0.9 ~ 0.1 and then change the behavior of the arrows to go by 0.1 increments 
## Any other notes & comments?
[image]

tatokis, 2022-05-03:
Could set an adaptive StepType for Qt >= 5.12 https://doc.qt.io/qt-5/qdoublespinbox.html#setStepType and for < 5.12 call `setSingleStep()`.

I think 0.1 increments might be a bit too low, so 0.2 might be a good compromise.

I'll leave it open as it should be really simple to do for someone who wants to contribute, but if no one gets around to doing it, I'll do it some time in the future.

bharath000, 2022-08-02:
Hi tatokis, Can I work on this issue, I am new to the project.

tatokis, 2022-08-02:
@bharath000 Sure thing

jsaarela1, 2022-09-21:
Hi, could I give this issue a try? This will be my first contribution but I'd like to help.

BlakeOneal, 2022-09-23:
Hey, I'd also like to work on this issue if possible. I should have a solution done by this weekend, let me know if that's alright!

bharath000, 2022-09-25:
Hi Folks, 

Sry for delay in response, Please go ahead and work on this.