ckb-next/ckb-next #824: Mouse not functional after waking up from suspend; infinite loop retrying bragi device mode

doubleplusc, 2022-05-03:
## General information

### Source
How and where did you get this program, e.g.:
aur/ckb-next-git 1:0.4.4.r201.gcb37c692-1
Issue also present in manual install based on b07541d 

## Logs & crash reports
[ckb-next-report-1651595978.gz](https://github.com/ckb-next/ckb-next/files/8612170/ckb-next-report-1651595978.gz)

## What is the program's current behaviour?
If I suspend the computer, I can wake it by moving the mouse. However, the lights don't turn on, and the cursor does not move.
If I try restarting the daemon, the lights turn on, but the cursor still does not do anything.
Disconnecting and reconnecting the mouse works.
You can see from the logs that the daemon gets into an infinite loop retrying to get bragi device mode and timing out.

## What is the expected behaviour of the program?
The mouse should be fully functional after waking up from a suspend without needing to reconnect USB cable.

Thanks!

tatokis, 2022-05-25:
Can you try the code from the [bragi_wakeup_fix](https://github.com/ckb-next/ckb-next/compare/bragi_wakeup_fix) branch? https://github.com/ckb-next/ckb-next/commit/79c2c70294e777a202398bcd12a37d398076ab70 should fix it.

closed 2022-05-27 by tatokis

doubleplusc, 2022-05-27:
Can confirm the issue is now fixed. Thanks!

commit 6c5a72f0 (2022-05-27): Merge branch 'bragi_wakeup_fix'

Closes #824