ckb-next/ckb-next #828: Corsair M65 RGB Elite don't work on mousepad

d3m0-sm, 2022-05-11:
I use a M65 RGB Elite and have installes the newest version of cbk-next on Kubuntu 20.04.

If i use my M65 on my mousepad, the corsair won't move after login. The Optical Sensor works just fine on Windows. 

In DirectX the mouse is shown with the drivers of cbk-next

ovandermeer, 2022-05-14:
I'm having the same issues with my M65 RGB Elite, and a temporary fix seems to be setting the "lift height" to the highest option. You'll need to connect another (non-corsair) mouse to your pc, because as far as I can tell there's no easy way to change the setting without one. To change the mouse height, launch the ckb-next software, go to the M65 RGB Elite tab, go to "Performance", and change "Lift height" in the bottom right to "High". Unfortunately this setting flips back whenever you change modes or create a new mode, so it's not an ideal solution.

tatokis, 2022-05-23:
How many Lift Height options does this mouse have in iCUE?

d3m0-sm, 2022-07-10:
Thx ovandermeer. Thas solved the Problem

closed 2024-05-02 by gravityfargo