ckb-next/ckb-next #829: event17...client bug: event processing lagging behind by ##ms, your system is too slow

mutte-79, 2022-05-12:
## General information
I get this constantly repeating log entry.
The weird thing is that it does not matter if I use the 1ms poll mode, 30FPS along with a rainbow animation  or if I use the 8ms poll mode, 5FPS and no animations at all, the supposed lag stays around 15-30ms.
I don't experience any lag, don't se any flickering or other issues.
I tested connecting to both USB 1 and USB 2 ports. (Though I don't think the keyboard supports USB 2)
And I have tried connecting the keyboard connector to USB1 & USB2 and the light connector to an external power source. What ever I do I cant get rid of that log entry and I can't change the lag time.

Log Entry: "_event17 - ckb1: Corsair K70 RGB Gaming Keyboard vKB: client bug: event processing lagging behind by 23ms, your system is too slow_"

### Source
I have built manually, installed pre-built and installing from repositorys, always the same result.

## What is the expected behaviour of the program?
To behave like a gentleman and keep it's opinion about my equipment to it self.

[ckb-next-report-1652395487.gz](https://github.com/ckb-next/ckb-next/files/8683126/ckb-next-report-1652395487.gz)

Ravenslofty, 2022-05-13:
That warning is actually coming from libinput not ckb-next; the only reason ckb-next is mentioned is because it's where the input events come from.

closed 2022-05-13 by Ravenslofty

mutte-79, 2022-05-14:
Oh ok, I'm sorry, my bad.
Thanks for your answer. I'll poke around and see if I can manage to disable that logging.

zDEFz, 2022-09-15:
I tested this with a K100 RGB and I found out this issue only appears when ckb-next is installed. What is the reason? I like the K100-RGB, but I want to disable the lighting altogether... What can I do?