ckb-next/ckb-next #831: kperfwidget.ui typo prevents compile

spmorton, 2022-05-16:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

### Source
How and where did you get this program, e.g.:
  * downloaded zip file from github on 5/16/2022
  * Built manually using quickinstall script
  

## Logs & crash reports

AutoUic subprocess error
------------------------
The uic process failed to compile
  "SRC:/src/gui/kperfwidget.ui"
into
  "SRC:/build/src/gui/ckb-next_autogen/include/ui_kperfwidget.h"
included by
  "SRC:/src/gui/kperfwidget.cpp"

Compile error
-------
/usr/lib/qt5/bin/uic -o /home/scott/Downloads/ckb-next-master/build/src/gui/ckb-next_autogen/include/ui_kperfwidget.h /home/scott/Downloads/ckb-next-master/src/gui/kperfwidget.ui

Output
------
uic: Error in line 940, column 40 : Unexpected element item
File '/home/scott/Downloads/ckb-next-master/src/gui/kperfwidget.ui' is not valid

gmake[2]: *** [src/gui/CMakeFiles/ckb-next_autogen.dir/build.make:77: src/gui/CMakeFiles/ckb-next_autogen] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:900: src/gui/CMakeFiles/ckb-next_autogen.dir/all] Error 2
gmake: *** [Makefile:171: all] Error 2
An error occurred,
press enter to exit.

## Any other notes & comments?
line 939 has 
4/item> should be </item>

tatokis, 2022-05-17:
There is no `4/item>` anywhere in that file, and said file has not been modified for the past 3 years.
https://github.com/ckb-next/ckb-next/blob/7443761c7784a25a0692c19ab1d14062a74fa54c/src/gui/kperfwidget.ui#L937-L941

Looking at those those characters in binary:
```
4: 0b00110100
<: 0b00111100
```
Sounds like you had a single bit flip somewhere in memory.
Might be time to run memtest86+ on your computer, or check your storage medium for corruption.

As for how to "fix" this, delete your local copy and redownload the repo.

closed 2022-05-17 by tatokis

spmorton, 2022-05-17:
it had to happen during the download, cracked it open directly from the
archive and the flip is there. haven't seen anything like that in a long
time.

Thanks for the quick response.