ckb-next/ckb-next #837: Keyboard not responding anymore when wake from suspend

pakro, 2022-05-30:
Hi, I've searched the problem and there some older reports which look familiar to my issue. There are all solved however. This problem was introduced for me with the 0.5.0 release. The master branch doesn't work also. Maybe this is also related to Arch and some update they made. If so I beg you pardon.

You can simply reproduce this by putting the PC into suspend and wake (waking with keyboard works) it. Afterwards the keyboard does not react anymore until I plug the USB out and in again. 

I'll also attach the report the ckb-next program:
[ckb-next-report-1653902822.gz](https://github.com/ckb-next/ckb-next/files/8797239/ckb-next-report-1653902822.gz)

pakro, 2022-05-30:
Yes from what I can tell this solves the problem on my machine. At first it didn't help, because I didn't reboot. But after building it from source without the mentioned commit and rebooting the machine afterwards, I now can successfully suspend and wake without having to plug in and out the USB cable of the keyboard after waking.

tatokis, 2022-06-01:
This is extremely frustrating... This fixes another suspend issue that I could actually reproduce.
The "proper" solution is much more complicated, but I'll try to get to it soon. Apologies

pakro, 2022-06-01:
No need for apologies. I can live with the workaround for the time being.
Just in that moment I also realized that that this workaround also has it flaws, because isn't working right now. I need to replug keyboard too. So the old solution isn't bulletproof either, it just tends to work better with my setup.
And nobody else seems to have troubles with the new implementation or I am the one of the few people who suspend their PC most of the time. Either way I just wait for the other solution. If you have more urgent topics on which you need to work just to them first.
So no rush please.

pakro, 2022-06-03:
So out of interest I've tried the 0.5.0 stable version again because the workaround also had its problems. This time I rebooted after installing and now I don't have any problems with suspend -- at least the short suspends work without a problem but last time I checked even the short ones didn't work without reattaching the keyboard. So I'm not sure if this a valid bug any longer. Or any other issue or special arrangement with my other hardware.

Anyone else can reproduce this? If not, I would suggest to close it. I beg you pardon for any effort I have unnecessary caused.

gctypo, 2022-06-05:
I've had this issue (probably the same, but potentially different); I don't need to replug my keyboard (K70 Lux), but it does take ~30sec to be responsive.

Reverting the commit brought that down to ~10sec before being responsive after wake from suspend, but there's still some delay.

pakro, 2022-06-15:
@gctypo Although I must admit I'm not someone having a huge patience I think the issues are different because I'm very sure that at some point when suspending and waking the PC again I've waited longer than 30 secs and it still didn't respond without a re-plug.

skjerns, 2023-01-13:
I think I have the same problem, however it appears only every once in a while. I have not figured out what the circumstances are. Is it working atm for you guys?

Glich440, 2023-07-08:
Same issue on latest version with the K60

skjerns, 2023-07-09:
> Same issue on latest version with the K60, undoing the commit no longer seems to work

which version are you using? My ppa was using an old version, installing from master solved my problem

Glich440, 2023-07-09:
latest from AUR ckb-next 0.6.0-20-g932ff43

Glich440, 2023-07-09:
> > Same issue on latest version with the K60, undoing the commit no longer seems to work
> 
> which version are you using? My ppa was using an old version, installing from master solved my problem

latest from AUR ckb-next 0.6.0-20-g932ff43
but I have tried manualy compiled from source and it also seems to fail.