ckb-next/ckb-next #838: Corsair K55 RGB PRO

tzymni, 2022-05-30:
Corsair K55 RGB PRO is not detected.  It's possible to add this support in the future? 

ckb-next-dev-detect - Mon May 30 17:34:55 CEST 2022

```
Kernel: Linux tomasz-pop-os 5.17.5-76051705-generic #202204271406~1653440576~21.10~6277a18 SMP PREEMPT Wed May 25 01 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Pop
Description:	Pop!_OS 21.10
Release:	21.10
Codename:	impish

USB Devices:

Bus 001 Device 007: ID 1b1c:1ba4 Corsair CORSAIR K55 RGB PRO Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1ba4 
  bcdDevice            1.02
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K55 RGB PRO Gaming Keyboard
  iSerial                 3 F5001BC7603D610AAF8A0D2410030023
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      82
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
[... 240 lines cut ...]
	libFLAC.so.8 => /lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007f7bf1853000)
	libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f7bf1825000)
	libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f7bf1778000)
	libopus.so.0 => /lib/x86_64-linux-gnu/libopus.so.0 (0x00007f7bf171a000)
	libogg.so.0 => /lib/x86_64-linux-gnu/libogg.so.0 (0x00007f7bf170f000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f7bf16e4000)
	liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f7bf16c4000)
	libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x00007f7bf16b9000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f7bf1596000)
	libnsl.so.2 => /lib/x86_64-linux-gnu/libnsl.so.2 (0x00007f7bf157a000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f7bf1575000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f7bf1567000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f7bf153f000)
	libtirpc.so.3 => /lib/x86_64-linux-gnu/libtirpc.so.3 (0x00007f7bf1512000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f7bf14ed000)

Environment:
QT_ACCESSIBILITY=1
XDG_SESSION_DESKTOP=pop
XDG_SESSION_TYPE=x11
QT_BEARER_POLL_TIMEOUT=-1
XDG_CURRENT_DESKTOP=pop:GNOME
DISPLAY=:1
QT_IM_MODULE=ibus
```

gravityfargo, 2022-06-27:
The K55 has been marked as supported since MAR2018 according to a[ Supported Hardware](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware/64649a0d0ab32cd3ffc2602b6bb5b2ef029c579a) page commit, but there are three different models of K55 that aren't specified in our current documentation. Can you give me the SKU of your keyboard so I can make note of this? It's on the [sticker underneath](https://help.corsair.com/hc/en-us/articles/360025378691-Find-a-Serial-Number-or-Lot-Code).

Ravenslofty, 2022-06-27:
> Hello.
>
> Sure, my serial number is 029522109110 . Will you add this support in the
> future?

That's not an SKU, and we don't have the Corsair magic to turn serial
numbers into SKUs anyway.

Could you run `lsusb -vd 1b1c:` and paste the output for me?

tzymni, 2022-06-27:
<pre>
Bus 001 Device 004: ID 1b1c:1ba4 Corsair CORSAIR K55 RGB PRO Gaming Keyboard
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1ba4 
  bcdDevice            1.02
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K55 RGB PRO Gaming Keyboard
  iSerial                 3 F5001BC7603D610AAF8A0D2410030023
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      82
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

</pre>

gravityfargo, 2022-06-27:
Sorry, I'm not a dev or actively contributing to device support. Just trying to help with documentation. There are 3 variants of K55. PRO, PRO-LITE, and PRO XT. Current documentation doesn't specifically say which K55 is supported, just that one is. Is yours specifically the "K55 PRO" variant according to the sticker underneath? I'm not sure if the lsusb product is accurate to the specific model.

tzymni, 2022-06-27:
On the sticker underneath its 'CORSAIR K55 RGB PRO Gaming" . Model RGP0118.

Flums, 2022-07-07:
Daemon logs:
`juli 07 09:56:18 desktop ckb-next-daemon[8895]: [I] Connecting CORSAIR K55 RGB PRO Gaming Keyboard at /dev/input/ckb1
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [I] Starting input thread for /dev/input/ckb1
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-8/1-8:1.0/ep_81
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-8/1-8:1.1/ep_82
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-8/1-8:1.2/ep_83
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:18 desktop ckb-next-daemon[8895]: [I] Attempting reset...
juli 07 09:56:19 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:19 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:20 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:20 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:20 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:20 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:21 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:21 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:21 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:21 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:22 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:22 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:22 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:22 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:23 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:23 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:24 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:24 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:24 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:24 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:25 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:25 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:26 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:26 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:26 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:26 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:27 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:27 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:27 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
juli 07 09:56:27 desktop ckb-next-daemon[8895]: [F] start_bragi_common (device_bragi.c:106): ckb1: Couldn't get bragi device mode. Aborting
juli 07 09:56:28 desktop ckb-next-daemon[8895]: [E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[... cut ...]
juli 07 09:56:32 desktop ckb-next-daemon[8895]: [I] Disconnecting /dev/input/ckb1
juli 07 09:56:32 desktop ckb-next-daemon[8895]: [I] Caught internal signal SIGUSR2
juli 07 09:56:32 desktop ckb-next-daemon[8895]: [I] Stopping input thread for /dev/input/ckb1
juli 07 09:56:32 desktop ckb-next-daemon[8895]: [E] os_usb_control (via input.c:530):  No such device, res = 0xffffffff
juli 07 09:56:32 desktop ckb-next-daemon[8895]: [I] Stopping indicator thread for ckb1 (19)
juli 07 09:56:32 desktop ckb-next-daemon[8895]: [I] Removed device path /dev/input/ckb1
`

Flums, 2022-07-07:
[image]

Flums, 2022-07-07:
[image]

wolframIII, 2022-07-12:
Bus 005 Device 002: ID 1b1c:1ba4 Corsair CORSAIR K55 RGB PRO Gaming Keyboard
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1ba4 
  bcdDevice            1.01
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K55 RGB PRO Gaming Keyboard
  iSerial                 3 F5001BC35FC4654CAF7D910102021020
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      82
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

tatokis, 2022-07-12:
There's a branch you can test out https://github.com/ckb-next/ckb-next/tree/k55pro. It needs some changes before it can be merged, but it should be functional.

tzymni, 2022-08-17:
> There's a branch you can test out https://github.com/ckb-next/ckb-next/tree/k55pro. It needs some changes before it can be merged, but it should be functional.

I can confirm that in this branch, everything seems to work correctly. Thank you!

closed 2022-08-18 by tatokis