ckb-next/ckb-next #842: MM700 RGB Gaming Mousepad

pakro, 2022-06-18:
I would kindly ask if there is someone out there who could add support for the MM700 RGB Gaming Mousepad.

The USB ID is **1b1c:1b9b**

Thank you! 
---

Enter the name of your device here and attach the report generated above.

gravityfargo, 2022-10-05:
Added to the supported devices wiki as "Requested"

prutheus, 2023-05-17:
What is the current state for the MM700RGB? Will it be supported soon?

CedricB31, 2023-12-24:
Hello all,

More info about MM700RGB mousepad, i have sniff USB package with my analyzer (Beagle USB Protocol Anlayser). **It implement the bragi protocol**. As the picture shows, there is effectively 3 RGB zones that can be set with a BRAGI_WRITE_DATA command (Frame example:
` 0x08 0x06 0x00 0x09 0x00 0x00 0x00 R_zone1 R_zone2 R_zone3 G_zone1 G_zone2 G_zone3 B_zone1 B_zone2 G_zone3`.

This table below show an extract of my sniffer package at the first connection the RGB color of every zone are 0xFFFFFF (full sniffing in [test.csv](https://github.com/ckb-next/ckb-next/files/13761484/test.csv))  

| HIDReport | Data |  Comment |
| --------- | ---------------------------------------------------------------------------------- | --------- |
| Out | 08 02 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | read VID |
| In  | 00 02 00 1C 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | 0x1B1C |
| Out | 08 02 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | read PID |
| In  | 00 02 00 9B 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | 0x1B9B |
| Out | 08 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | read BARGI MODE|
| In  | 00 02 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | HARDWARE mode |
| Out | 08 02 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | read APP Version |
| In  | 00 02 00 01 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | V1.0.5 |
| Out | 08 02 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | read bld version |
| In  | 00 02 00 01 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | v1.0.3 |
| Out | 08 01 03 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | set BARGI MODE to soft  |
| In  | 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | resp |
| Out | 08 02 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| In  | 00 02 00 1C 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| Out | 08 02 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| In  | 00 02 00 9B 1B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| Out | 08 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| In  | 00 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| Out | 08 02 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| In  | 00 02 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| Out | 08 02 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | read BARGI MODE |
| In  | 00 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | SOFT mode |
| Out | 08 02 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| In  | 00 02 00 01 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| Out | 08 02 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| In  | 00 02 00 01 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | |
| Out | 08 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Read BRIGHTNESS |
| In  | 00 02 00 17 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Resp 0x0317 79.1% |
| Out | 08 02 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Read BRIGHTNESS|
| In  | 00 02 00 17 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Resp 0x0317 79.1% |
| Out | 08 0D 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Open handle|
| In  | 00 0D 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Resp |
| Out | 08 06 00 09 00 00 00 FF FF FF FF FF FF FF FF FF 00 00 00 00 ... | SetColor Z1:White Z2:White Z3:white |
| In  | 00 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... | Resp |

I will try to integrate the support of this MM700RGB mouse pad in the ckb-next-daemon, and if you need more trace I can do more.

CedricB31, 2024-04-07:
Hi @pakro and @prutheus,
I have added the MM700 support in my fork. My PR to integrate this new device is open #1050, but if you want to try you this support before integration you can clone my fork (branch mm700 https://github.com/CedricB31/ckb-next/tree/mm700). And if you see any issue, don't hesitate to let me know.

CedricB31, 2025-03-24:
Hello @gravityfargo,

The MM700 support has been added in [v0.6.2](https://github.com/ckb-next/ckb-next/releases/tag/v0.6.2). 
Maybe you can update the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware#other-devices) and close this issue.

closed 2025-05-13 by gravityfargo