ckb-next/ckb-next #844: Nightsword "Profile Down" button still does not work when rebound

Kamoedesu, 2022-06-23:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

### Source
How and where did you get this program, e.g.:
  * Package from AUR/PPA/COPR/...
  * Built manually using quickinstall script
  * Pre-built dmg for macOS

Built manually with quickinstall script.

## Logs & crash reports
[ckb-next-report-1656023066.gz](https://github.com/ckb-next/ckb-next/files/8971747/ckb-next-report-1656023066.gz)

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?

The Profile Down button does not register inputs when the binding is changed from it's default setting. Resetting the button back to it's default binding (Profile Down) works, but changing it's binding to anything else (F10 in my case) will not allow it to work.

## What is the expected behaviour of the program?

All buttons should input their default binding and rebound inputs properly.

## Any other notes & comments?

I previously had version 0.4.4 installed, I uninstalled it in order to install the newest version. This fixed the Profile Up button, but the Profile Down button remains broken as before.

As I'm a very infrequent Linux user, I'll be willing to offer any information I can, but I'll likely need very involved instructions on how to troubleshoot due to my low skill level.

sfinseth, 2023-05-14:
I am also running into this issue, is there any way I can help provide any information or details that might help resolve this? Or any help on a jumping-off point in order to try and contribute a fix myself?

Eir-nya, 2024-12-17:
Same here. I have ckb-next 0.6.0-1.3 installed via the AUR. When ckb-next is not running, the "profile up" and "profile down" buttons on my wired nightsword mouse function as they normally would. When ckb-next is running, I can rebind both buttons, but "profile down" is not detected no matter what I bind it to. Every other button works perfectly.

Eir-nya, 2024-12-17:
I did some digging and found #701 which appears extremely similar but at the time affected both profile up and profile down buttons.

Following the instructions from [this comment](https://github.com/ckb-next/ckb-next/issues/701#issuecomment-950950591), I got this output:

No GUI
```
Profile up button
[I] ckb1 Input: (81) 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Profile down button
[I] ckb1 Input: (81) 03 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
With GUI
```
Profile up button
[I] ckb1 Input: (81) 03 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Profile down button
[I] ckb1 Input: (81) 03 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[I] ckb1 Input: (81) 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

The values I got were the same with and without the GUI, I tried multiple times. I don't think I did anything wrong here?

The keycodes I got seem to be the same as [this comment](https://github.com/ckb-next/ckb-next/pull/656#issuecomment-729265011) on the PR that first added Nightsword support. Hopefully someone with more knowledge than me can use this information to fix the issue!

Yueh72, 2025-03-31:
I got the same thing with this button. It seems CKB has problem with the mices.