ckb-next/ckb-next #848: Issue Clean-up

gravityfargo, 2022-06-26:
Hello, me again :) 

**Hardware requested by issue is now listed as "Experimental" or is now supported**
 - #74 
 - #101
 - #346
 - #505 
 - #520 
 - #543
 - #558 
 - #575
 - #669 
 - #698
 - #716 
 - #751 
 - (may be a bug disguised as a new-device request) #838

**Informative Posts are probably best for a wiki/closure due to age or relevancy**
 - #52 

**Issue fixed but not closed**
- #249 this was mentioned in #215 and was [supposedly](https://github.com/ckb-next/ckb-next/issues/215#issuecomment-499246530) pushed to master and marked as closed. there was an unrelated issue brought up after the issue was closed and as a result, reopened. Seems to be related to a system resume/wakeup bug making the keyboard unresponsive. See the section below that groups those (probably not dupes, but most likely connected.)

**Possible dups/same issue diff hardware mentioning sleep or hibernation or suspends - possibly systematic not device-specific**
A new label may be helpful here.
- #230 
- #249 
- #256 
- #456 
- #476 
- #545 
- #652 
- #693 
- #727 

**macOS isn't supported anymore**
- #310 
- #364 
- #431 
- #446 
- #478 
- #480 
- #482 
- #484 
- #488 
- #504 
- #512 
- #516 
- #527 
- #529 
- #534 
- #598

Ravenslofty, 2022-06-28:
> **Device requested is now listed as "Experimental" or is now supported**

These are left open **because** they're listed as experimental; it's better to have a single thread for each device until it's completely known to work.

> **Informative Posts are probably best for a wiki/closure due to age or relevancy**

It was an issue because issues can be pinned.

> **Issue fixed but not closed**

These are...very much closed issues?

> **Possible dups/same issue diff hardware mentioning sleep or hibernation or suspends - possibly systematic not device-specific** A new label may be helpful here.

Perhaps they are representative of a systematic thing, but we can't prove that, so unless/until we can get somebody to show it's not, we tend to assume so. Corsair's firmware is exceptionally buggy and working around it is something we've had to do in the past.

> **macOS isn't supported anymore**

These were all already closed except for the latter which only mentions macOS in its context of blocking support for some kind of CI, which is an entirely separate thing.

gravityfargo, 2022-06-28:
I was unaware of the policy to keep device requests open through the experimental stage, good to know! Maybe a tag for the future?

The "Fixed but not closed" and macOS issues were open when I made this issue. "KitsuWhooa" in the IRC closed them all yesterday.

I noticed the pattern of issues with sleep/hibernation and wanted to point out the number of issues with similar problems. I didn't mean to imply they should be closed.

Thank you for the clarifications : )

closed 2022-06-28 by gravityfargo