ckb-next/ckb-next #853: please consider discord server

Derik-for-real24, 2022-06-30:
Hey,

Could you guys please consider to make a discord server regarding your rgb projects for corsair. I recently found out about your work, and I was hoping to get my corsair rgb fans working on linux, but your projects needs really more attention, I think Discord is a good place to share your work nd vision easily nd it should be more accessable for more people.

I like to ask question nd see where I can help, im not a programmer but I do have some insight.

tatokis, 2022-06-30:
No.

closed 2022-06-30 by tatokis

Derik-for-real24, 2022-06-30:
really, why would that be, any issues with discord ?

steinex, 2022-06-30:
There are many reasons for why Discord is a bad choice for open source projects. It's proprietary, you are locked to one client, it collects data about you, just to name a few.

If you are interested in a Discord-like experience while chatting with us, consider using Matrix with the Cinny client.

Derik-for-real24, 2022-06-30:
yeah thats my goal, i like to talk with you guys nd learn something nd also help with the project where is possible.

Im not familiar with Matrix or Cinny client, are those apps supported on windows, cuz im on windows currently, is it also free to use ?

Derik-for-real24, 2022-06-30:
thank you so kindly, I will do that nd hopefully I can join your community without issues.