ckb-next/ckb-next #854: Corsair Dark Core RGB Pro SE: Mouse clicks do not work

daugustin, 2022-07-06:
## General information
When I start ckb-next-daemon, mouse clicks do not work anymore. Mouse Movement still works. As soon as I stop the daemon, clicks work instantly again.

I have this:
Corsair CORSAIR DARK CORE RGB PRO SE Gaming Dongle

The mouse works fine under Linux. When I start the daemon with the needed --enable-experimental mouse movement still works, but mouse clicks are gone.

### Source
ckb-0.5.0 from here https://github.com/gentoo/gentoo/pull/26109

## Logs & crash reports
As I cant reach the settings menu with an attached device, I can only show the logs by the daemon (started on the shell to get the logs easily):

```
# /usr/bin/ckb-next-daemon --enable-experimental
ckb-next: Corsair RGB driver 0.5.0
[I] You have enabled support for experimental devices in a ckb-next stable release. It is recommended to use the latest code from git.
[I] Root controller ready at /dev/input/ckb0
[I] Adding device with vendor = 0x1b1c, product = 0x1b7f, path = /dev/bus/usb/001/003, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7
[E] patchkeys (keymap_patch.c:130): Could not find profswitch in base keymap. Not patching profswitch for mouse.
[I] Connecting CORSAIR DARK CORE RGB PRO SE Gaming Dongle at /dev/input/ckb1
[I] Claiming interfaces. name=CORSAIR DARK CORE RGB PRO SE Gaming Dongle, firmware=0504, ep_str= 6
[I] ckb1: Claiming 6 interfaces
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/ep_82
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/ep_82 has wMaxPacketSize 64
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/ep_84
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/ep_84 has wMaxPacketSize 64
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.2/ep_83
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.2/ep_83 has wMaxPacketSize 64
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.3/ep_81
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.3/ep_81 has wMaxPacketSize 64
[I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.4/ep_85
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.4/ep_85 has wMaxPacketSize 64
[I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.5/ep_86
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.5/ep_86 has wMaxPacketSize 64
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb1: Pairing id: 00 00 00 00 00 00 00 00 
[I] Setup finished for /dev/input/ckb1
[E] bragi_open_handle (bragi_common.c:193): Failed to open handle 0x0. Error was 0x1
[E] setactive_bragi (device_bragi.c:86): ckb1: Bragi light init returned error 0x1
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[W] bragi_set_property (bragi_common.c:30): Failed to set property 0x2. Device said it's not supported.
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[W] bragi_set_property (bragi_common.c:30): Failed to set property 0x2. Device said it's not supported.
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[E] bragi_led_count (led_bragi.c:39): Unknown product 0x1b7f
[... Repeating endlessly]
```

tatokis, 2022-07-06:
Two things:

> [I] You have enabled support for experimental devices in a ckb-next stable release. It is recommended to use the latest code from git.

And have a look at https://github.com/ckb-next/ckb-next/pull/840

Clofriwen, 2022-07-07:
Hello,

New pull opened at this place : https://github.com/ckb-next/ckb-next/pull/855
- that made from an current master version
- support an corrected battery status (improved compatibility with nxp protocole (Dark Core))

closed 2024-05-02 by gravityfargo