ckb-next/ckb-next #861: ckb-next packaged in Alpine Linux

stacyharper, 2022-07-18:
Hey there,

We packaged ckb-next in aports testing tree. It already is available to edge user.

https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/36302

Could you update the wiki ?

The users have to :

- Add testing/ tree in their apk repository list (if not already)
- `apk add ckb-next ckb-next-daemon` (we separated both)
- `rc-service ckb-next-daemon start` and `rc-update add ckb-next-daemon` (to start and enable the openrc service)

tatokis, 2022-10-05:
Really sorry for the super late response. Is the first step still necessary? (As in, has it made to "stable", or whatever the equivalent is in alpine?)

Thank you very much for packaging!

stacyharper, 2022-10-05:
Np for the delay.

Yes the first step is necessary but it is nothing
uncomon. Until it land on the community tree user need to enable
testing.

gravityfargo, 2024-05-02:
[Added](https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#alpine-linux)

closed 2024-05-02 by gravityfargo