ckb-next/ckb-next #863: Allow selection from a range of colors in animations

RedKnight26, 2022-07-30:
## What is the program's current behaviour?
Each animation (though I'll be specifically talking about ripple type lighting) only lets you select one color for the animation.

## What would you like it to do instead?
I would like animations to be able to randomly select one color from a preset range of colors so that when each key is pressed it will emit a different colored ripple.

## Any other notes & comments?
I have the attention span of a goldfish and when I'm bored I like watching the pretty colors ripple out on each key press :)

tatokis, 2022-07-30:
Isn't that what the "Randomly select from gradient" option does?
[image]

RedKnight26, 2022-08-08:
> 

That works, but I can't pick rainbow for my gradient. Is there a way to do that?

tatokis, 2022-08-10:
I'm not sure I understand. Why can't you pick a rainbow for the gradient? There should be a rainbow preset to select.

RedKnight26, 2022-08-11:
> I'm not sure I understand. Why can't you pick a rainbow for the gradient? There should be a rainbow preset to select.

I think it has something to do with my monitor size. When I click on the gradient, everything's incredibly hard to see. This happens semi-frequently due to my 55" 4k TV
[image]

RedKnight26, 2022-08-11:
quick update, I got it! Though I wish the ripples would be a single color and not actual rainbow ripples

RedKnight26, 2022-08-11:
update, I got it working!