ckb-next/ckb-next #865: K70 RGB Pro

SmoochySix4, 2022-07-31:
K70 RGB Pro
[ckb-next-report-1659234171.gz](https://github.com/ckb-next/ckb-next/files/9227911/ckb-next-report-1659234171.gz)

SmoochySix4, 2022-07-31:
Got the newest version from the PPA and still not supported. Here is the report from that version...
[ckb-next-report-1659234703.gz](https://github.com/ckb-next/ckb-next/files/9227937/ckb-next-report-1659234703.gz)
.

Theymoon, 2022-08-09:
Here is another report of the K70 RGB PRO. There is a message "No device connected" in the GUI. The keyboard is, however, listed in the report file.

[ckb-next-report-1660029044.gz](https://github.com/ckb-next/ckb-next/files/9288167/ckb-next-report-1660029044.gz)

Beyley, 2022-08-23:
Not sure if it falls under the same device, but heres my report of the K70 RGB PRO MINI WIRELESS
[ckb-next-report-1661295501.gz](https://github.com/ckb-next/ckb-next/files/9411021/ckb-next-report-1661295501.gz)

gravityfargo, 2022-10-05:
@Beyley I don't see the RGB PRO Mini in the supported list, and no previous issues seem to request it. You should submit a new issue with that report to request support. @Theymoon

PCMartinzzon, 2023-08-22:
[ckb-next-report-1692722560.gz](https://github.com/ckb-next/ckb-next/files/12410847/ckb-next-report-1692722560.gz)
Updated report if it helps =)

PCMartinzzon, 2023-08-22:
Updated report if it helps =)
[ckb-next-report-1692723018.gz](https://github.com/ckb-next/ckb-next/files/12410891/ckb-next-report-1692723018.gz)

jasha23, 2023-08-27:
ID for mine is also 1bd4

[ckb-next-report-1693148053.gz](https://github.com/ckb-next/ckb-next/files/12448240/ckb-next-report-1693148053.gz)

gravityfargo, 2023-09-18:
@jasha23 it looks like 1bd4 is a separate variant. Could you open a new issue with your report and mention I asked you to open it?

SmoochySix4, 2023-09-24:
Still doesn't seem to be working. Here is the report from the git version.

[ckb-next-report-1695538039.gz](https://github.com/ckb-next/ckb-next/files/12707981/ckb-next-report-1695538039.gz)

KarkanAlzwayed, 2023-11-25:
I just got this keyboard and it doesn't even show on the app. How can I get the model number on it so I can post it here? Thanks

ErwinJobse, 2023-12-13:
[ckb-next-report-1702494157.gz](https://github.com/ckb-next/ckb-next/files/13664625/ckb-next-report-1702494157.gz)

RokeJulianLockhart, 2023-12-25:
What PR fixed this?

jayamorin, 2024-04-03:
Here's mine.
[ckb-next-report-1712159218.gz](https://github.com/ckb-next/ckb-next/files/14855361/ckb-next-report-1712159218.gz)

gravityfargo, 2024-05-01:
I am reopening this issue,  the device request can be considered open since that PR has not been merged yet, and this issue is referenced on the wiki.

WhateverFits, 2024-05-13:
I also have 1bc4. I attempted to use the PR 1030 modifying the identifier and that broke the keyboard communications so it appears to be a different variant.

gravityfargo, 2024-05-13:
@Beyley 
`1bb6`  is a different device and has not been requested yet, please open a separate issue.

@kyleguarco @jasha23
`1bd4`  is a different device and has its own request. #996

@PCMartinzzon @ErwinJobse  @WhateverFits 
`1bc4` is a different device and has not been requested yet, please open a separate issue.

flashmaddison, 2024-09-18:
Using the latest version ckb-next 0.6.0-49-g7708b0c (from https://launchpad.net/~tatokis/+archive/ubuntu/ckb-next-git)

"No devices connected"

[ckb-next-report-1726644538.gz](https://github.com/user-attachments/files/17040279/ckb-next-report-1726644538.gz)

usernamecopied31, 2024-10-02:
Using the latest version and same issue product id 1bb3.

[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/17236085/ckb-next-dev-detect-report.gz)

MathisDupriez, 2024-11-11:
Using the version 0.6.0 on fédora 41, i see id 1bb3, still not working

commit 86cbf83c (2025-03-17): add support for k70 pro

closes #865
PR #1030 (2025-03-17): add support for k70 pro
Adds support for the K70 RGB Pro (1bb3)

~All the LEDs are working except the logo light. If there are ideas about how to get the logo light working I'm happy to make the change and test~ Fixed with @JohnoKing's patch.

closes #865
closes #915 
closes #928 
closes #998
closed 2025-03-17 by tatokis