ckb-next/ckb-next #868: K55 PRO XT: WinLock Led Indicator do not work

hlechner, 2022-08-04:
## General information

### Source
Installed using AUR ([ckb-next-git](https://aur.archlinux.org/packages/ckb-next-git))

## Logs & crash reports

Attach:
[ckb-next-report-1659640667.gz](https://github.com/ckb-next/ckb-next/files/9262698/ckb-next-report-1659640667.gz)

```
$ zcat ckb-next-report-1659640667.gz | grep Failed | awk ' BEGIN { first = 8; last = 20 }
{ for (i = first; i < last; i++) { printf("%s ", $i) } print $last }' | sort | uniq
```

result:
```
(bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.  
(bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.  
(bragi_common.c:13): Failed to get property 0x96. Device said it's not supported.  
(bragi_common.c:30): Failed to set property 0x2. Device said it's not supported.  
```

## What is the program's current behaviour?
When **WinLock button** is pressed it does work by disabling/enabling the windows key, however the non-RGB led indicator do not light when the WinLock is enabled.

## What is the expected behaviour of the program?
When the WinLock is enabled, the non-RGB led indicator should light, as it works in hw mode.

## Any other notes & comments?
It do not light the non-RGB led indicator but it does shows on the application:
[image]
[image]

tatokis, 2022-08-11:
What if you change this
https://github.com/ckb-next/ckb-next/blob/6c5744262a194435ae0f386c430ed780522d1243/src/daemon/keymap.c#L420
to
```
{ "lock",            114, KEY_CORSAIR },
```

hlechner, 2022-08-11:
> What if you change this
> 
> https://github.com/ckb-next/ckb-next/blob/6c5744262a194435ae0f386c430ed780522d1243/src/daemon/keymap.c#L420
> 
> 
> to
> ```
> { "lock",            114, KEY_CORSAIR },
> ```

I tried this and also the numbers: `115`,`116`,`117`,`118`,`119` but the led didn't light.

If there is a way I can diagnose or further help, just let me know how

tatokis, 2023-03-29:
@hlechner If you are still around and you have time, can you please try the number 0 instead?

As in:
```
{ "lock",            114, KEY_CORSAIR },
```

I'm fairly confident the winlock indicator is in fact RGB.

Also, make sure to test without https://github.com/ckb-next/ckb-next/pull/869 as that disables the lock LED, and as recently discovered for another K55, if you disable it, then there will be no winlock led control at all. (I'll get to that PR soon hopefully, sorry for the _extremely_ late response.)

tatokis, 2023-04-01:
Try changing the winlock colour in the performance tab and see what happens.

hlechner, 2023-04-01:
Here my test:
[image]

I tried with a few colors on that and it doesn't changed the LED color but the brightness of it. As example:
White (full brightness) > Green > Blue > Red (the less brightness compared to others)

The colors I used were the colors in the picker of GTK (that ones on screenshot).

closed 2023-04-01 by tatokis

hlechner, 2023-04-01:
@tatokis 
I tested the commit https://github.com/ckb-next/ckb-next/commit/1a855b1b557d7530ce78f0c4dacd2d7b25d5ea6c but it's not working.

Right now the button stopped working (not just the indicator), if deselect "Indicate Windows Lock" under performance tab the LED instantly turns on (but do not reflect the lock state, it just stay on).

I also tried to change the line 420 of ``keymap.c`` to ``{ "lock",            0, KEY_CORSAIR },``  (as I did before the commit) but now it doesn't have effect

hlechner, 2023-04-01:
Find out:

keymap_patch.c:
```c
// Just winlock
static const keypatch k55proxtpatch[] = {
    {  96,  "lock",    0, KEY_CORSAIR },
};
```

I replace the number ``96`` to ``114`` and it start to work.

commit 1a855b1b (2023-04-01): Fix winlock for K55 RGB Pro XT

Closes #868