ckb-next/ckb-next #874: Break off QuaZip/Zip from ckb-next-daemon executable (out of scope)

egberts, 2022-08-26:
## General information

Wow.  I've been propelled to 370+ revisions via (`git pull`) later and it no longer compiles.  I'm  on Debian 11 latest and my selection of Window Manager is immaterial here.

Given my 22 years of vulnerability research and as a defensive secure coding architect, I went donning on my code review hat and dove into the latest ckb-next code.

I became horrified to learn that firmware interaction was dovetailed into the general CKB-NEXT daemon.  

Plus some perturbation of the fact that a sound support has been added NOT as a separate daemon.

I am sadden that this wonderful project (that I helped) has gone far too off my rail (of keeping non-GUI and GUI separate).  I worked with `virt-manager` (GUI-based) and `virsh.sh` (CLI) using `libvirt`; this ckb-next can still be salvage to follow this model of separation (and they are really good at that).

### Source

August 22, 2022 Git tree (both v0.5.0 and v0.4.4)

## Logs & crash reports

Not applicable

## What is the program's current behaviour?

Jack-of-all-trade, master of all vulnerabilities.

## What is the expected behaviour of the program?

ckb-next-daemon should be only about the state of the LEDs.  Disengage and breakout all other functionalities (including flash files).

A separate library libckb-next should provide the interfaces with both text console and GUI-based terminal.

A separate music visualizer is its own daemon via interaction with a new `libckb-next.so`.

A separate one-shot daemon to test firmware for applicable upgrade.  Not everyone wants this.

## Any other notes & comments?

Not sure if anyone has experienced the excellent `libvirt`/`virt-manager` reference.

egberts, 2022-08-26:
Reverting back to eef033f  (Nov 20, 2022).

egberts, 2022-08-27:
Then, I shall submit additional but concise issues that I am currently experiencing.

closed 2022-08-27 by egberts

egberts, 2022-08-27:
I think I found the NXP but not the NXP 5 protocol specs.

https://www.nxp.com/docs/en/user-guide/UM10204.pdf

where would one find a Bragi spec?

Ravenslofty, 2022-08-27:
That's not the NXP protocol we mean; for one it's an I2C protocol, whereas this is USB. We call it the NXP protocol because all the devices that use it use NXP microcontrollers (and we also believe it to have been originally based on some NXP demonstration code).

I wrote [an incomplete description](https://github.com/ckb-next/ckb-next/wiki/Corsair-Protocol) of the NXP protocol version 4 (before Bragi we called it "version 2" after the firmware revisions, but we now call it NXP version 4 after the `P04` USB string), but in general both protocols are proprietary and are entirely undocumented, let alone having prescriptive specifications.