ckb-next/ckb-next #876: Arabic numbers in DPI selection

AnzeG8, 2022-09-06:
All the number selections in Performance tab for my Corsair Scimitar Elite are using Arabic numbers. I'm guessing it's reading the serial code of the mouse and determined Middle East origin. Please add a language check to this, if the interface language is set to English there's no reason to have Arabic take over on the numbers, regardless of serial ID origin location. Thank you.

Running Mint Cinnamon 20.3, latest ckb-next release manual build with git clone and quickinstall.
[ckb-next-report-1662457675.gz](https://github.com/ckb-next/ckb-next/files/9495530/ckb-next-report-1662457675.gz)

Ravenslofty, 2022-09-06:
Could you send a screenshot?

AnzeG8, 2022-09-06:
Nevermind, please ignore me, my system region and datetime settings reverted back to Middle East and I didn't notice. Uninstalled, changed system settings, restart, rebuild ckb-next and I'm on normal numbers now.
I really should have checked that as a first, my bad.
Anyway, thank you for your work on this, it's appreciated.

closed 2022-09-06 by AnzeG8