ckb-next/ckb-next #880: Can I use this with runit?

xylorp, 2022-09-20:
Hi, I'm just wondering if it would be possible to use this with the runit init system. Forgive me, but to your knowledge: 

If I were to put the ckb-next-daemon in /var/run/runit/service/, would I be able to run it? 

Thank you!

tatokis, 2022-09-24:
I do not know how runit works but I think you should be able to generate a sysvinit file using cmake ( or manually edit https://github.com/ckb-next/ckb-next/blob/master/linux/sysvinit/ckb-next-daemon.in) and put that in `/etc/init.d/` without the `.in` extension, or however one would install init.d scripts.

That said, you should contact user support channels for your distribution and ask your question there, or read the documentation.

closed 2022-09-24 by tatokis

steinex, 2022-09-25:
Yes, you can use the daemon with runit. See the service file that Void Linux ships for example: https://github.com/void-linux/void-packages/blob/master/srcpkgs/ckb-next/files/ckb-next-daemon/run - it's as easy as that.