ckb-next/ckb-next #882: [Regression] Delay on key lighting effects when typing fast

tippl, 2022-09-21:
## General information

### Source
  * Package from AUR
  * Manually built for testing

## Logs & crash reports

[ckb-next-report-1663792891.gz](https://github.com/ckb-next/ckb-next/files/9620224/ckb-next-report-1663792891.gz)

## What is the program's current behaviour?

I use the `Type Lighting` animation, key turns white on press, on release slowly turns back default color.
After recent update, when typing fast, i'm experiencing a delay on this animation.

To put it simply, the animation can't keep up. After a few words, the keys turning white are a few seconds behind. Even fully lifting hands off keyboard, the keys keep lighting up for a few seconds until they catch up.

Repro should be easy, setup the `Type Lighting` animation and type really fast, try to hit a lot of different keys and not mash same keys at the same time. Keep it up for a bit and the delay should add up to be noticeable.

## What is the expected behaviour of the program?

Expected behavior is that it keeps up no matter how fast i type.

## Any other notes & comments?

The issue happens on current master branch, installing stable version fixes the issue.

I have bisected the issue leading to this commit: f54f4ab35092c961353e91baaa7cdd86a0c39974

```
f54f4ab35092c961353e91baaa7cdd86a0c39974 is the first bad commit
commit f54f4ab35092c961353e91baaa7cdd86a0c39974
Author: Tasos Sahanidis <tasos@tasossah.com>
Date:   Mon Sep 12 10:37:53 2022 +0300

    Rewrite command fifo reader

 src/daemon/command.c |  26 ++++---------
 src/daemon/command.h |   2 +-
 src/daemon/devnode.c | 106 +++++++++++++++++++--------------------------------
 src/daemon/devnode.h |  11 ++++--
 src/daemon/usb.c     |  88 ++++++++++++------------------------------
 5 files changed, 80 insertions(+), 153 deletions(-)
```

tatokis, 2022-09-23:
Thank you very much for reporting this. Unfortunately the issue doesn't seem to be with that commit specifically, but rather the changes in it brought up a few fundamental issues in how packets are sent.

As a temporary workaround while I come up with a proper fix, with the latest code from master, after making sure that the FPS in the GUI is set to **30** (and with the GUI running), you can run the following in a terminal (the command is supposed to have 60, yes):
```bash
bash -c 'echo fps 60 | tee /dev/input/ckb{1..9}/cmd 2> /dev/null'
```

After a few seconds, the keyboard will catch up and the animations will run in realtime again.

tippl, 2022-09-25:
Followed the workaround and can't say it worked for me. Reinstalled from current master, checked GUI has FPS set to 30, ran the command and everything seems to be as if i didn't run anything.

I'll just install stable and wait for a proper fix.

Grocel, 2022-09-28:
I have a similar issue of the effect being "one key press behind" on recent builds. I'm not sure if this is related.

closed 2022-09-29 by tatokis

tatokis, 2022-09-29:
This should be fixed now. Part of the issue was that the new implementation prevented missed frames from being skipped, which I've now fixed.

Another part is that ckb-next-daemon is throttling the rgb frames incorrectly, so for example setting it to 30Hz actually sends frames at ~28Hz, or 60Hz ends up being 50Hz.
Because of this, the old frames would get queued up, and since with that change that caused the regression the delayed frames weren't being dropped, they just built up and all of the animation would fall behind until the pipe filled up eventually.

I'm working on fixing the refresh rate issue now too.

commit 00b11738 (2022-09-27): Remove FPS command and lower usbsend delays

Devices now actually run at the set framerate.

1300us delay was picked because it's the only way to reach 60FPS.
If some devices are unhappy about it, then the FPS command will need
to be reintroduced.

Closes #882
commit 7304323c (2022-10-01): Allow multiple command lines to be processed at once

This prevents delayed rgb commands from endlessly building up until
the pipe fills up and stalls.

Also fixes frames being delayed due to f54f4ab3509.

Closes #882