ckb-next/ckb-next #883: Support left handed mouse toggle, like Windows icue has

RichardJECooke, 2022-09-22:
## What is the program's current behaviour?
There is no setting for the mouse to toggle between left-handed and right-handed. icue in Windows has this.

This setting is essential because many games (and possible other graphics apps) access the mouse buttons directly, and don't use your operating system's left or right handed setting. It's the whole reason I bought a Corsair M55 RGB Pro mouse in the first place.

## Any other notes & comments?
Maybe it's here somewhere and I'm not seeing it? Or perhaps you have a code snippet  you can send me that will work on the command line? :)

Thanks!

tatokis, 2022-09-23:
There's no easy one click checkbox, and unfortunately unless someone adds it, it'll be a while as there are more important things. For now, you can simply rebind all the buttons and swap them as you wish. It's a bit annoying, but it can be done.

For example, you can swap the left and right clicks, and unbind the left buttons and bind their functionality on the right ones. Though be sure to change the right click to left click first, otherwise you won't be able to select things anymore unless you switch profiles/modes.

RichardJECooke, 2022-09-23:
Perfect, thank you! Swapping two bindings worked perfectly.