ckb-next/ckb-next #884: ckb-next branch k100-v2 knob issue

zDEFz, 2022-09-25:
**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

=> Error creating textual authentication agent: Error opening current controlling terminal for the process (`/dev/tty'): No such device or address

## General information

### Source
How and where did you get this program, e.g.:
  * Branch k100-v2

## Logs & crash reports
Not possible to gather

## What is the program's current behaviour?
When I use the knob above ESC and F1, it prints out 
'bcmnopqrstuvwxyz1234567890['
It keeps then spamming the '['

## What is the expected behaviour of the program?
If not configured, do nothing.

## Any other notes & comments?

Check required #878  Added functionality for K100 Mechanical #878

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. Please make sure to upload a report when submitting an issue.

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo

mmtmn, 2025-06-24:
re-open issue

tatokis, 2025-06-24:
Yeah, sorry, this issue should not have been closed

mmtmn, 2025-06-29:
I've added a fix for it over at #1170

Ergwano, 2025-09-22:
It would be great getting this resolved since as cat owner having the wheel unintentionally triggered is pretty common. A bunch of inputs triggered in some sensitive locations is dangerous and basically makes this software unusable for me.

Ergwano, 2025-09-26:
@tatokis Ah I see, that is good to hear. I am using 0.6.2 from the arch repo at the moment. I'll go try the git version from aur then. Is there a rough time frame when there will be a release build again?

Thank you for the reply as well!