ckb-next/ckb-next #886: iCue Nexus support

Jimbo7136, 2022-09-28:
Can you please, please, please look into better options for the iCue Nexus? The options corsair has are lackluster at best.

Thank you.

gravityfargo, 2022-10-05:
No one has requested support for this device yet, so this issue should be labeled as a new device request. Could you upload please a report?

Jimbo7136, 2022-10-05:
Not sure what you mean by report.

I must apologize, though. When I submitted this request I did so because I found an open source alternative to corsair icue software, but I didn't realize you don't support windows.

I don't use icue features in linux, only server stuff.

christophermichaelshaw, 2022-11-04:
Here is a system report with the nexus screen attached to my K70RGB MK2

Jimbo7136, 2022-11-04:
I'm going to ask you to create a new thread and close this one.

I don't want to be emailed every time there's an update on this. I should have closed this thread weeks ago.

closed 2022-11-04 by Jimbo7136