ckb-next/ckb-next #887: Option to completely disable the management of the RGB

eSPiYa, 2022-09-28:
## What is the program's current behaviour?
Once the application is started, it will override the RGB control from OpenRGB. And OpenRGB can't find my keyboard anymore.

## What would you like it to do instead?
Give the user an option to disable the management of RGB and let the other applications to do it instead.

## Any other notes & comments?
I installed this application so I can assign macros on the special keys found in K95 RGB and Scimitar. I prefer OpenRGB to manage the RGB of my computer because of wider support on different devices which enables users to sync everything regardless of brand.

tatokis, 2022-09-28:
As it stands, it's not really viable due to how the devices function.

Maybe complain to the OpenRGB devs. If they were being cooperative instead of "being inspired" by the work from this project while trying to replace it, and not contributing anything back, I'd have been willing to work on integration with them.

closed 2022-09-28 by tatokis

eSPiYa, 2022-09-28:
> Maybe complain to the OpenRGB devs. If they were being cooperative instead of "being inspired" by the work from this project while trying to replace it, and not contributing anything back, I'd have been willing to work on integrating with their project.

Looks like there's some kind of personal issue between you and the OpenRGB dev(s). Based on OpenRGB's statement, it seems like there's an issue with libusb that CKB-Next is using that locks up the device and made it not detectable from OpenRGB. I went back to Windows temporarily to check if the same issue happens with iCue but OpenRGB can detect the Corsair devices and can temporarily interfere with it. 

So, OpenRGB's statement most probably is true and the feature I'm asking for is theoretically achievable. Also people would prefer something that can properly sync their RGBs between different brands.

Ravenslofty, 2022-09-28:
Okay, no, the problem is not libusb, and it's not ckb-next, and it's not OpenRGB. The problem is USB, or more specifically, USB-related design choices that Corsair have made.

Let me explain: A USB device communicates through "endpoints"; think of them as like TCP or UDP ports on a website. When your computer (the "host") communicates to your keyboard (the "device"), it also communicates to a specific endpoint. Here's the thing: unlike TCP, there's no way of keeping track of who a request is associated with, so when a device replies, it has no idea who to send the reply to. That's why Linux only allows one process access to a USB endpoint.

Now, Corsair devices (at least, modern ones) only have two endpoints: one for writing data and one for reading data . This "data" includes keypresses, profile data, RGB, macro keys, and other things. Trying to work around this by e.g. only writing RGB data to the write endpoint, and only reading keypresses from the read endpoints runs into problems: you can't do anything which requires the device to respond, like *initialise the device* (ever seen a packet dump in the ckb-next-daemon log that starts with `0e 01`? that's a handshake packet).

So between the user having a keyboard that works, and one that supports multiple applications trying to use it, we picked the former. If the OpenRGB devs dislike our decision, they can either reimplement the keypress and input layer like we did, or they can instead use one of the APIs that we provide specifically so that other applications can draw animations.

eSPiYa, 2022-09-28:
>Here's the thing: unlike TCP, there's no way of keeping track of who a request is associated with, so when a device replies, it has no idea who to send the reply to. That's why Linux only allows one process access to a USB endpoint.
I see, got this part. But I assume here that you use the read endpoint to capture not only the keypress but also responses when you send something to the device through the write endpoint that is why you CKB-Next hogged the connection. This sounds like an asynchronous request. But how critical is the response if you send something to the device like changing the color or saving a macros? Or changing something requires a series of back and forth communication between the host and the device?