ckb-next/ckb-next #891: Corsair M65 RGB Elite (dev ID 1b5a)

atomGit, 2022-10-04:
i see ckb-next can apparently write to hardware memory for both the M65 RGB (1b12) and Pro (1b2e), but not the Elite (1b5a)

i assume the Elite uses a different protocol?

tatokis, 2022-10-04:
Does it support hardware animations? As in, do the LEDs change colours when there's no software/ckb-next isn't running? If so, yeah, it's a different format that has not been reverse engineered.

gravityfargo, 2022-10-05:
For future reference to other people, later on, this issue is about hardware playback. The M65 RGB Elite is a supported device.

atomGit, 2022-10-05:
supported, yes, but not in the context of my question

the Elite is supported as long as the ckb-next daemon is running, but not (currently) supported as far as ckb-next being able to write a config to mouse memory

gravityfargo, 2022-10-09:
"this issue is about hardware playback"