ckb-next/ckb-next #895: Music visualisation doesn't work with Ubuntu 22.04

AFCMS, 2022-10-10:
## General information

### Source

ppa:tatokis/ckb-next-git

## Logs & crash reports

[ckb-next-report-1665430732.gz](https://github.com/ckb-next/ckb-next/files/9748510/ckb-next-report-1665430732.gz)

## What is the program's current behaviour?

When I follow the wiki guide, it doesn't work:

[image]

[image]

[image]

Then I play music (using rythmbox or vlc), the keyboard stays black (as the preview it self), despite pavucontrol showing the blue bar moving.

## What is the expected behaviour of the program?

Show a nice music RGB visualisation :/

## Any other notes & comments?

It worked just fine with Ubuntu 20.04.

Maybe yet another problem with Wayland?

tatokis, 2022-10-11:
What if you delete the existing animation and add a new "Spectrum (Monitor)" one? With the latest changes, it's supposed to just work out of the box without changing anything in pavucontrol.

AFCMS, 2022-10-11:
Yes I tried it and this didn't work as expected.

But, I messed in the settings and with a 100 sensibility and a 150% volume in pavucontrol, the animation works (with any modes).

IDK why, because with the default settings it worked perfectly on Ubuntu 20.04.

tatokis, 2022-10-13:
Does system/sound card volume affect it somehow?
Is the volume of your music player set high enough? This would happen if the volume, for example on Rhythmbox, is really low.

Ideally this would be solved by implementing AGC in the visualiser.

AFCMS, 2022-10-13:
The sound level in Rhythmbox is ~50%, but same problem with VLC at 100%

The listening level is super good, both with and without earphones.