ckb-next/ckb-next #896: Update Supported Devices Wiki

gravityfargo, 2022-10-10:
- added #865 request and device id.
- added device id for #842
- added request for #886
- converted all of the "Requested #000" to hyperlinks

[Patch in a zip](https://github.com/ckb-next/ckb-next/files/9749653/0001-added-K70-RGB-Pro-request-and-device-id.zip)

gravityfargo, 2022-10-29:
- added #900 request and device id.

You can [view my gist](https://gist.github.com/gravityfargo/2fbce071791a4ea015b27a33a4abf534) to see a live preview.

I also made a [tutorial with photos](https://gist.github.com/gravityfargo/f136425487767dd620d737d76162825c) to enable experimental devices via systemctl that I think I forgot to post about. 

**Patch**
[0001-Added-devices-updated-links-updated-requests.tar.gz](https://github.com/ckb-next/ckb-next/files/9892633/0001-Added-devices-updated-links-updated-requests.tar.gz)

gravityfargo, 2022-11-14:
- added #907 
- updated #906 

You can [view my gist](https://gist.github.com/gravityfargo/2fbce071791a4ea015b27a33a4abf534) to see a live preview.

[changes.tar.gz](https://github.com/ckb-next/ckb-next/files/9999323/changes.tar.gz)

rokejulianlockhart, 2023-02-10:
Why not submit as a pull request?

gravityfargo, 2023-02-12:
The wiki can only be modified by contributors, I can clone it but there is no way to fork it so I've got to make patches for a contributor to apply manually.

closed 2023-04-19 by gravityfargo