ckb-next/ckb-next #900: CORSAIR VOID ELITE Wireless Gaming Headset

reyx, 2022-10-22:
### What kind of device is it?

Headphone

### What is the name of the device?

CORSAIR VOID ELITE Wireless Gaming Headset

### Link to Corsair's Product Page

https://www.corsair.com/br/en/Categories/Products/Gaming-Headsets/USB-Headsets/VOID-RGB-ELITE-USB-Premium-Gaming-Headset-with-7-1-Surround-Sound/p/CA-9011204-NA

### Additional Details

Add any additional details here.

### Generated Report

ckb-next-dev-detect - Sat Oct 22 16:58:25 -03 2022

```
Kernel: Linux fedora 5.19.16-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Oct 16 22:50:04 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

OS Version:

USB Devices:

Bus 003 Device 024: ID 1b1c:0a50 Corsair CORSAIR VOID ELITE Wireless Gaming Headset
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0a50 
  bcdDevice            0.00
  iManufacturer           2 Corsair
  iProduct                3 CORSAIR VOID ELITE Wireless Gaming Headset
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              1 Hid Interface
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
[... 404 lines cut ...]
	libgsm.so.1 => /lib64/libgsm.so.1 (0x00007fbfcdfde000)
	libFLAC.so.8 => /lib64/libFLAC.so.8 (0x00007fbfcdf83000)
	libvorbis.so.0 => /lib64/libvorbis.so.0 (0x00007fbfcdf53000)
	libvorbisenc.so.2 => /lib64/libvorbisenc.so.2 (0x00007fbfcdea8000)
	libopus.so.0 => /lib64/libopus.so.0 (0x00007fbfcde4a000)
	libogg.so.0 => /lib64/libogg.so.0 (0x00007fbfcde3f000)
	libmpg123.so.0 => /lib64/libmpg123.so.0 (0x00007fbfcdde2000)
	libmp3lame.so.0 => /lib64/libmp3lame.so.0 (0x00007fbfcdd6a000)
	libbz2.so.1 => /lib64/libbz2.so.1 (0x00007fbfcdd57000)
	libbrotlidec.so.1 => /lib64/libbrotlidec.so.1 (0x00007fbfcdd4a000)
	libgpg-error.so.0 => /lib64/libgpg-error.so.0 (0x00007fbfcdd22000)
	libpcre2-8.so.0 => /lib64/libpcre2-8.so.0 (0x00007fbfcdc85000)
	libbrotlicommon.so.1 => /lib64/libbrotlicommon.so.1 (0x00007fbfcdc62000)

Environment:
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=wayland
QT_BEARER_POLL_TIMEOUT=-1
XDG_CURRENT_DESKTOP=GNOME
WAYLAND_DISPLAY=wayland-0
GNOME_SETUP_DISPLAY=:1
QT_WAYLAND_DECORATION=gnome
DISPLAY=:0
QT_IM_MODULE=ibus
```

joker-119, 2024-03-19:
Year and a half later, is there any update on if this is even ever going to be looked into?

t3kg33k, 2024-04-14:
> Year and a half later, is there any update on if this is even ever going to be looked into?

What do you mean? Was this now included in recent updates being that the above mentioned are closed?
The reason I ask is because I was kind of wondering the same thing. I'm pretty sure I have the most recent update and I have this headset but it is not displayed as connected hardware in the ckb-next applet. I see that it is in the supported hardware list but it isn't working for me.

t3kg33k, 2024-04-14:
I guess I didn't look close enough at the Device ID in the supported hardware. Mine is the following:

Bus 002 Device 002: ID 1b1c:**0a51** Corsair CORSAIR VOID ELITE Wireless Gaming Dongle

Will 0a51 ever be included?

gravityfargo, 2024-05-02:
FFR I edited OP's post to put the generated report in a collapse section.

mattmattox, 2024-06-10:
Just an FYI, I was able to get LED and Battery/Charging status working in Ubuntu 24.04 using the following:

https://github.com/Sapd/HeadsetControl  <<== CLI (Backend)
https://extensions.gnome.org/extension/5823/headsetcontrol/  <<== Gnome extension (Frontend)

Note: You might need to edit the permissions of /dev/hidraw5 (chown username:username /dev/hidraw5) so that headsetcontrol works without sudo.

t3kg33k, 2025-03-10:
I didn't see an answer to my question.

1b1c:0a51 Corsair CORSAIR VOID ELITE Wireless Gaming Dongle

Will 0a51 ever be included?

milgner, 2025-03-31:
Strange: mine seems to have a different ID again...

```
ID 1b1c:0a55 Corsair CORSAIR VOID ELITE Wireless Gaming Dongle
```