ckb-next/ckb-next #902: Support for Portuguese ISO Layout on Linux

gabrielnoal, 2022-11-01:
### Enhancement Details

Hi, I just bought a K55 RGB PRO in Portugal and I'm having a big problem to use it in Portuguese.

I found [this issue](https://github.com/mattanger/ckb-next/issues/165) from the old repository, and besides the OS that I'm using, linux with KDE Plasma distrubution, it's the same problem.

In the issue thread, there is already a solution by @MauricioFauth , dont't know if it works for this new version, but would help me a lot if someone could add it.

MauricioFauth, 2022-11-01:
I added support for the ABNT2 standard, which is the standard used for Brazilian Portuguese keyboards. What is the standard/keyboard layout used in Portugal?

gabrielnoal, 2022-11-01:
Hi @MauricioFauth, i dont know.
I'm brazilian too, but could find ABNT2 layout too.

[image]

[image]

MauricioFauth, 2022-11-01:
That's because your keyboard doesn't use the ABNT2 keyboard layout. See: https://pt.wikipedia.org/wiki/Teclado_(inform%C3%A1tica)#Disposi%C3%A7%C3%A3o_do_teclado_em_Portugal_e_no_Brasil

MauricioFauth, 2022-11-01:
I'm not familiar with the current state of the code base, but I don't think it's too different to implement the portuguese layout than what I did for the ABNT2 layout in https://github.com/mattanger/ckb-next/pull/331.

gabrielnoal, 2022-11-01:
So my keyboard uses ABNT not ABNT2 correct?

gabrielnoal, 2022-11-01:
Looking at the keyboard settings, it was in portuguese BR, but when I changed to PT, the keys started to match. The only problem its that in ckb-sofware still not matching, because it doesn't has the portuguese layout.

MauricioFauth, 2022-11-01:
You keyboard layout looks more like the ISO layout.
See: https://commons.wikimedia.org/wiki/File:Physical_keyboard_layouts_comparison_ANSI_ISO_KS_ABNT_JIS.png

MauricioFauth, 2022-11-01:
> Looking at the keyboard settings on my pc, it was in portuguese BR, but when I changed to PT, the keys started to match. The only problem its that in ckb-sofware still not matching, because it doesn't has the portuguese layout.

You probably have to add a patch for PT_PT keymap. For example:

https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/gui/keymap.h#L106

https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/gui/keymap.cpp#L96-L101

https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/gui/keymap.cpp#L604-L606

gabrielnoal, 2022-11-03:
@MauricioFauth , I could add everythink besides de Portuguese patch, could you help me with that?

closed 2024-05-02 by gravityfargo