ckb-next/ckb-next #906: Corsair iCUE NEXUS Companion Touch Screen

christophermichaelshaw, 2022-11-05:
### What kind of device is it?

Keyboard

### What is the name of the device?

iCUE NEXUS Companion Touch Screen

### Link to Corsair's Product Page

https://www.corsair.com/us/en/Categories/Products/Accessories-%7C-Parts/CORSAIR-iCUE-NEXUS/p/CH-9910010-NA

### Additional Details

Happy to help with any further development -- I started to write some python code to push some basic output to the screen, but found ckb-next, which might be a better jumping off point.

Mine is connected to a Corsair K70RGB MK2 -- I have a couple of them, so I'm able to test with different firmwares, etc. Thanks! :)

Bus 001 Device 010: ID 1b1c:1b8e Corsair CORSAIR iCUE NEXUS
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b8e 
  bcdDevice            2.00
  iManufacturer           1 iDisplay Technology Co., Ltd.
  iProduct                2 CORSAIR iCUE NEXUS
  iSerial                 3 01FE210301806
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              288mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     245
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               1

### Generated Report

Drag your generated report here.
[ckb-next-report-1667539167.gz](https://github.com/ckb-next/ckb-next/files/9944368/ckb-next-report-1667539167.gz)

mattmattox, 2023-06-26:
It would really nice to add support for this device

tuc0w, 2024-02-23:
I'd love support for the Nexus too. Please make it happen 😢

DigitalDreamn, 2024-07-05:
I would also like to see support for the Nexus added.

Melk047, 2025-01-08:
just bought the nexus, i agree with this

Melk047, 2025-05-09:
https://github.com/jurkovic-nikola/OpenLinkHub

This supports the nexus, if any of you need an app to use it still