ckb-next/ckb-next #907: K100 Air Wireless Keyboard

chfritz, 2022-11-05:
### What kind of device is it?

Keyboard

### What is the name of the device?

K100 Air

### Link to Corsair's Product Page

https://www.corsair.com/us/en/Categories/Products/Gaming-Keyboards/K100-AIR-WIRELESS-RGB-Ultra-Thin-Mechanical-Gaming-Keyboard---CHERRY-MX-Ultra-Low-Profile-Tactile/p/CH-913A01U-NA

### Additional Details

I've already tried the https://github.com/ckb-next/ckb-next/tree/k100-v2 branch to see whether, by any chance, the K100 Air was behaving just like the K100, but it doesn't seem like it. The device Ids are different, of course, but even after I modified the device id in the code to fool ckb-next into treating my K100 Air like a K100 all I got was a recognized device but nothing worked: all lights on the keyboard were off and the keyboard didn't even function anymore -- key presses had no effect.

I'll try to provide a usb dump from wireshark once I figure out how to find the correct usbmon device to record.

### Generated Report

Drag your generated report here.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/9944401/ckb-next-dev-detect-report.gz)

gravityfargo, 2022-11-14:
Was the keyboard directly connected via USB or were you using slipstream?

chfritz, 2022-11-14:
I had both USB cable and slipstream plugged in, but I've just replaced the report with a new one where the slipstream received was not plugged in -- only the usb cable from the keyboard.

villwi, 2022-11-14:
Hi,

I also have the same keyboard and it would be great to have ckb-next support for it.
I'm willing to help, if there is something I can do and do some testing but please keep in mind that I am linux noob.

I'm on Ubuntu 20.04.5 LTS and my motherboard is Asus Rog Strix x299-e gaming.
I haven't installed any ckb-next version on my computer.
I just connected the keyboard directly via usb and it works as a regular keyboard.
I also connected the keyboard using slipstream and it worked as well.
I updated the keyboard and slipstream dongle firmware to the latest version in my Windows laptop.

boppy, 2022-11-14:
Same here. No success with the [K100-v2](https://github.com/ckb-next/ckb-next/tree/k100-v2) branch (cloned, quickinstall'ed, enabled experimentals). But willing to help where possible.

* EndeavourOS (Arch, btw. ^^)
* K100 Air with German Keyboard Layout
* Latest Firmware on dongle and keyboard
* dongle and cable connection available

chfritz, 2023-04-23:
I finally got around to installing wireshark inside my virtualbox and run two usbpcap captures: [k100-air-usbpcap.zip](https://github.com/ckb-next/ckb-next/files/11304480/k100-air-usbpcap.zip)

In both cases I'm starting the capture without the K100 air plugged into the virtual box machine and then plugging it in. In `icue-space445566` I'm also sending some color commands to the keyboard via icue, including setting the space key color to #445566. Not sure whether recognizable things like that are useful or not.

Please let me know if did this right or you need something else. Also happy to take next steps with this file if someone could point me to a guide on how to extract from this the values/settings we need to add a K100-air profile to ckb-next.

tobidelius, 2023-07-01:
What's the status on this?

MikeL270, 2023-09-27:
I just got my k100 Today. The wireless receiver is picked up by CKB-Next on the official release (AUR) with experimental mode enabled (I run experimental in order to support my Ironclaw Wireless). The device works fine tirelessly when CKB-Next is not running. When plugged in CKB-Next does not recognize the device but it is functional. Please let me know what I can provide to assist in supporting this device, I can run wire shark captures, dump firmware and all the sorts.

gilvbp, 2023-12-26:
I'm going to get mine on 12 January. Then I will work on this.

taksyon, 2024-04-14:
any updates on this?

stepanselyuk, 2024-07-16:
Yes, if K100 Air connected via USB the ckb-next (git master) does not recognize it, even with experimental support activated. If connected with USB dongle, then if recognizes the keyboard, but the keys seems messed up and selecting modes in ckb-next seems does not have any effect on RGB lighting. If then connect keyboard by USB cable, along with USB dongle, it still recognized and keys are fine, but RGB lighting not changing.

```
$ sudo ckb-next-daemon --enable-experimental
ckb-next-daemon 0.6.0-40-g63733a5
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR SLIPSTREAM WIRELESS USB Receiver at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.2/ep_83
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.3/ep_81
[I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.4/ep_85
[I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.5/ep_86
[I] ckb1: bragi dongle hotplug thread started
[I] Found new bragi subdevice 1
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[I] ckb2: Subkb vendor: 0x1b1c, product: 0x1bab
[E] bragi_wl_string (bragi_common.c:216): Unknown product 0x1bab
[I] ckb1: bragi dongle hotplug thread finished
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x2. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x44. Device said it's not supported.
[I] ckb1: Pairing id: e4 ab 52 ec 78 15 e7 96 
[I] Setup finished for /dev/input/ckb1
[E] os_usb_interrupt_out (via firmware.c:45): Bad file descriptor, res = 0xffffffff
[W] _start_dev (device.c:151): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb2
[I] ckb1: bragi dongle hotplug thread started
[I] ckb1: Bragi subdevice ckb2 disappeared
[I] Disconnecting /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
[I] ckb1: bragi dongle hotplug thread finished
[I] ckb1: bragi dongle hotplug thread started
[I] Found new bragi subdevice 1
[E] handle_bragi_key_input (keymap.c:609): Invalid length 15 and header 0x1 combination in handle_bragi_key_input()
[I] ckb2: Subkb vendor: 0x1b1c, product: 0x1bab
[E] bragi_wl_string (bragi_common.c:216): Unknown product 0x1bab
[I] ckb1: bragi dongle hotplug thread finished
[E] os_usb_interrupt_out (via firmware.c:45): Bad file descriptor, res = 0xffffffff
[W] _start_dev (device.c:151): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb2
-^[[17~[I] ckb1: bragi dongle hotplug thread started
[I] ckb1: Bragi subdevice ckb2 disappeared
[I] Disconnecting /dev/input/ckb2
[I] Removed device path /dev/input/ckb2
[I] ckb1: bragi dongle hotplug thread finished
^C
[I] Caught signal 2
[I] Disconnecting /dev/input/ckb1
[I] Caught internal signal SIGUSR2
[I] Stopping indicator thread for ckb1 (4)
[I] Stopping input thread for /dev/input/ckb1
[I] Removed device path /dev/input/ckb1
[I] Closing root controller
[I] Removed device path /dev/input/ckb0

```

RFL91, 2024-10-04:
anything new?

krzych-s, 2026-11-15:
[ckb-next-dev-detect-report.gz](https://github.com/user-attachments/files/34872943/ckb-next-dev-detect-report.gz)

Same here, please add support if possible... Very good keyboard
I can help with testing/debugging