ckb-next/ckb-next #909: Harpoon RGB Wireless - no mouse wheel scroll in experimental mode

drupalranger, 2022-11-18:
### Affected Devices

Harpoon RGB Wireless

### Linux Distribution

Ubuntu 22.04.1 LTS

### Installation Method

apt install ckb-next

### ckb-next version

ckb-next 0.5.0

### What did you expect?

I have a problem with mouse wheel, scroll does not work.
When running ckb-next in stable mode the device is not detected, 
so I've switched it to experimental mode using: 
```
sudo systemctl stop ckb-next-daemon
sudo ckb-next-daemon --enable-experimental
```
Here's the output of the commands:
```
ckb-next: Corsair RGB driver 0.5.0
[I] You have enabled support for experimental devices in a ckb-next stable release. It is recommended to use the latest code from git.
[I] Root controller ready at /dev/input/ckb0
[E] patchkeys (keymap_patch.c:130): Could not find profswitch in base keymap. Not patching profswitch for mouse.
[E] patchkeys (keymap_patch.c:130): Could not find profswitch in base keymap. Not patching profswitch for mouse.
[I] Connecting CORSAIR HARPOON RGB WIRELESS Gaming Mouse at /dev/input/ckb1
[I] Connecting CORSAIR HARPOON RGB WIRELESS Gaming Dongle at /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/ep_83
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb1: Pairing id: e0 62 0e 18 a6 e5 ce 1d 
[I] Setup finished for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb2
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/ep_83
[F] os_inputmain (usb_linux.c:164): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/ep_87, assuming 64
[I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.4/ep_85
[I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.5/ep_86
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb2: Pairing id: e0 62 0e 18 a6 e5 ce 1d 
[I] Setup finished for /dev/input/ckb2
```
In experimental mode the device is detected, I can access all settings and change following: 
- change RGB profiles 
- change buttons mapping - all buttons are working including wheel 

The only thing that doesn't work is wheel scroll, even though it seems to have proper binding configured. 

### Did you try anything that may have helped?

Things I've tried: 
- Reinstall the software 
- restart computer 
- plug/unplug the device 
- Look for similar issues in the queue 
- check troubleshooting / known issues guide 

### Generated Report

[ckb-next-report-1668790134.gz](https://github.com/ckb-next/ckb-next/files/10043209/ckb-next-report-1668790134.gz)

hockeymikey, 2023-05-10:
It's really strange, no scroll wheel action is registered. You can remap it and system doesn't pick up anything.

maorodribu, 2023-06-23:
I have the same problem, tested on linux deepin, ubuntu, fedora

olm52, 2023-08-01:
Just bought this mouse.
Same problem here

tatokis, 2023-08-09:
Can someone test the code from the [harpoon-wl-scroll](https://github.com/ckb-next/ckb-next/tree/harpoon-wl-scroll) branch?

drupalranger, 2023-08-10:
This seems to be working for me, 
I've installed the package with following commands: 
```

drupalranger, 2023-08-10:
This seems to be working, I've installed the package with following commands
```
git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
git checkout harpoon-wl-scroll 
git pull origin harpoon-wl-scroll
./quickinstall
```
Then I've restarted the deamon in experimental mode:
```
sudo systemctl stop ckb-next-daemon
sudo ckb-next-daemon --enable-experimental
```
Command output: 
```
$ sudo ckb-next-daemon --enable-experimental
ckb-next-daemon 0.6.0-30-ga2cc5cb
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR HARPOON RGB WIRELESS Gaming Mouse at /dev/input/ckb1
[I] Connecting CORSAIR HARPOON RGB WIRELESS Gaming Dongle at /dev/input/ckb2
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/ep_83
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb1: Pairing id: e0 62 0e 18 a6 e5 ce 1d 
[I] Setup finished for /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb2
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/ep_82
[I] Found EP 0x84 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/ep_84
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/ep_83
[F] os_inputmain (usb_linux.c:164): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.3/ep_87, assuming 64
[I] Found EP 0x85 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.4/ep_85
[I] Found EP 0x86 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.5/ep_86
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x2. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x44. Device said it's not supported.
[I] ckb2: Pairing id: e0 62 0e 18 a6 e5 ce 1d 
[I] Setup finished for /dev/input/ckb2
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
[W] bragi_process_notification (bragi_notification.c:36): ckb1: Unknown bragi notification for dev 1
```

I've been able to configure lightning profile and mouse wheel is now working without any extra configuration. 
If I try to run the package without experimental mode the device is not recognized. 
i'm currently using following ubuntu version:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.2 LTS
Release:	22.04
Codename:	jammy
```

drupalranger, 2023-08-10:
Generated report:
[ckb-next-report-1691651961.gz](https://github.com/ckb-next/ckb-next/files/12309904/ckb-next-report-1691651961.gz)

olm52, 2023-08-10:
Tested here and scroll is working in experimental mode.
DPI switch not working

drupalranger, 2023-08-10:
Indeed, DPI Switch button is not working , and also Binding panel is fully disabled , see screenshot
[image]

orosmatthew, 2024-01-20:
The harpoon-wl-scroll branch fixes the scrolling issue for me when in experimental. I'm on latest Arch.

fritz-fritz, 2024-03-25:
I have the same mouse and same issue with the current packaged version.

The harpoon-wl-scroll branch does indeed fix the scroll for me.

Additionally, for those having issue with the DPI button. It is mapped to the top left (up) button and can be enabled by binding to Change DPI > Cycle Up:
[image]

EatedPantsu, 2024-07-24:
With normal mode my mouse is not detected.

Using:

sudo ckb-next-daemon --enable-experimental

It is, and it seems everything works EXCEPT main point of mouse...movement. All buttons are registered as far as I can see and do work but just moving the cursor is not possible.

Using other mouse I was able to test changing colors etc work

Scroll does work.

Any help is appreciated and I'd give and logs/info that is needed.

ReemX, 2026-10-26:
Thanks @tatokis for the harpoon-wl-scroll branch. That fix (now in #1273, rebased onto v0.6.2) works perfectly on my Harpoon RGB Wireless (fw 5.0.129), both through the dongle and wired. Scroll, movement and buttons all work without `--enable-experimental`. It would be great to get #1273 merged so this can be closed.