ckb-next/ckb-next #911: In Settings and in Keyboard Bindings, allow the Hyper modifier

braoult, 2022-11-22:
### Enhancement Details

`Hyper` is a useful modifier, beside Shift, Control, Alt/Meta, and Super, and it would nice if we could assign a key to `Hyper Left` or `Hyper Right`.
**Note**: I can do it today via `xmodmap` or `xkb`, so adding this feature is more in a wish to have all settings in one place instead of multiple places...

### Additional Details

Add any additional details here.

JanNowak94, 2023-03-21:
@tatokis - recommended close; Reason: Hyper_L and Hyper_R are not input, but output of xkb - they don't occur as real keys in xkb/symbols (at least at time of this comment). So unless xkb changes (unlikely) this is not possible without additional xkb implementation. Which defeats purpose of this issue (unless someone writes entire xkb configuration for ckb "device" as packaged deal).

tatokis, 2023-03-21:
@JanNowak94 thanks!

For a workaround, maybe binding a macro that presses LCtrl LAlt and LSuper at the same time might get you a hyper. Alternatively, maybe xte keydown/keyup Hyper_L.

closed 2023-03-21 by tatokis

braoult, 2023-03-21:
My question was not to bind Hyper to some key (I know how to do that, and already do), but to have this "Hyper" in gui so that we can bind it to a key (add it in Settings menu, where we can already assign a key to `Caps/Shift/Control/Alt/Super`, `Hyper` is missing here).

tatokis, 2023-03-22:
ckb-next can only handle keys listed in [include/uapi/linux/input-event-codes.h](https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/input-event-codes.h) in the linux kernel. It works on a lower level than X and evdev/libinput. I do not see Hyper in there. There is only Meta, which is translated to the "Super" key.

If you can identify a specific key from there that isn't currently handled, but will be of help to you, let me know and I'll be happy to try to add it.

braoult, 2023-03-23:
I will (try to) make some tests this week-end to see if there is not simply a mismatch in events/key naming somewhere.
If `xkb-next` uses a lower level than X but has only 4 possible modifiers (`Shift`, `Control`, `Super`, `Alt`) while X has 8 - see below, I wonder how X could itself get those events. Maybe it accesses hardware a different way ?
X Modifiers (in <X11/X.h) (**Note: Comments beside "ModX" lines are mine**) :
```
/* modifier names.  Used to build a SetModifierMapping request or
   to read a GetModifierMapping request.  These correspond to the
   masks defined above. */
#define ShiftMapIndex           0        /* Shift */
#define LockMapIndex            1
#define ControlMapIndex         2        /* Control */
#define Mod1MapIndex            3        /* Alt / Meta */
#define Mod2MapIndex            4        /* Num Lock */
#define Mod3MapIndex            5        /* Hyper */
#define Mod4MapIndex            6        /* Super */
#define Mod5MapIndex            7        /* ISO Level3 Shift (Shift is level 2) - sometimes Alt Gr */
```
Maybe the "Menu" and "Function" (they look like more physical keys than modifiers for me) in `xkb-next` GUI are the same as missing X modifiers ? I am somehow lost here.

braoult, 2023-03-23:
> If I understand this correctly, they are handled as combinations of other keys. If I press Shift and Alt at the same time, I get a Meta in xev. Hyper is supposed to be Ctrl, Alt, Meta (Super), but it doesn't seem to work for me.

IMHO, this would not be useful, how would you combine `Hyper` with other modifiers ? `Ctrl` + `Meta/Alt` + `Super` (or Meta/Super in your notation), is seen as the combination of the 3 by applications. For example, we can use: `Control-Shift-Meta-Super-Hyper` + `a`, which is well understood by applications (5 modifiers + `a`). If a modifier was a combination of other modifiers, what would be the point ?