ckb-next/ckb-next #912: M65 Ultra Support

Ochinchin-san, 2022-11-23:
### What kind of device is it?

Mouse

### What is the name of the device?

M65 RGB ULTRA

### Link to Corsair's Product Page

https://www.corsair.com/de/en/Categories/Products/Gaming-Mice/FPS-Fast-Action-Mice/M65-RGB-ULTRA-Tunable-FPS-Gaming-Mouse/p/CH-9309411-EU2

### Generated Report

USB Devices:

Bus 003 Device 004: ID 1b1c:1b9e Corsair CORSAIR M65 RGB ULTRA Gaming Mouse
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b9e 
  bcdDevice            1.10
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR M65 RGB ULTRA Gaming Mouse
  iSerial                 3 1602B01BAB0521216195061DF5001906
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0074
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      76
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 111 lines cut ...]
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

Dmesg on current boot:
Nov 23 21:31:33.775751 ochin kernel: usb 3-4: New USB device found, idVendor=1b1c, idProduct=1b9e, bcdDevice= 1.10
Nov 23 21:31:33.775929 ochin kernel: usb 3-4: Product: CORSAIR M65 RGB ULTRA Gaming Mouse
Nov 23 21:31:33.776017 ochin kernel: usb 3-4: Manufacturer: Corsair
Nov 23 21:31:33.776117 ochin kernel: input: Corsair CORSAIR M65 RGB ULTRA Gaming Mouse as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-4/3-4:1.0/0003:1B1C:1B9E.0004/input/input4
Nov 23 21:31:33.776225 ochin kernel: hid-generic 0003:1B1C:1B9E.0004: input,hidraw3: USB HID v1.11 Mouse [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input0
Nov 23 21:31:33.776333 ochin kernel: hid-generic 0003:1B1C:1B9E.0005: hiddev97,hidraw4: USB HID v1.11 Device [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input1
Nov 23 21:31:33.776441 ochin kernel: hid-generic 0003:1B1C:1B9E.0006: hiddev98,hidraw5: USB HID v1.11 Device [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input2
Nov 23 21:31:33.776449 ochin kernel: input: Corsair CORSAIR M65 RGB ULTRA Gaming Mouse Keyboard as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-4/3-4:1.3/0003:1B1C:1B9E.0007/input/input5
Nov 23 21:31:33.776557 ochin kernel: hid-generic 0003:1B1C:1B9E.0007: input,hidraw6: USB HID v1.11 Keyboard [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input3

Dmesg on last boot:
Nov 23 16:59:14.805950 ochin kernel: usb 3-4: New USB device found, idVendor=1b1c, idProduct=1b9e, bcdDevice= 1.10
Nov 23 16:59:14.806139 ochin kernel: usb 3-4: Product: CORSAIR M65 RGB ULTRA Gaming Mouse
Nov 23 16:59:14.806233 ochin kernel: usb 3-4: Manufacturer: Corsair
Nov 23 16:59:14.806334 ochin kernel: input: Corsair CORSAIR M65 RGB ULTRA Gaming Mouse as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-4/3-4:1.0/0003:1B1C:1B9E.0004/input/input4
Nov 23 16:59:14.806451 ochin kernel: hid-generic 0003:1B1C:1B9E.0004: input,hidraw3: USB HID v1.11 Mouse [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input0
Nov 23 16:59:14.806567 ochin kernel: hid-generic 0003:1B1C:1B9E.0005: hiddev97,hidraw4: USB HID v1.11 Device [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input1
Nov 23 16:59:14.806683 ochin kernel: hid-generic 0003:1B1C:1B9E.0006: hiddev98,hidraw5: USB HID v1.11 Device [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input2
Nov 23 16:59:14.806692 ochin kernel: input: Corsair CORSAIR M65 RGB ULTRA Gaming Mouse Keyboard as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:06:00.3/usb3/3-4/3-4:1.3/0003:1B1C:1B9E.0007/input/input5
Nov 23 16:59:14.806805 ochin kernel: hid-generic 0003:1B1C:1B9E.0007: input,hidraw6: USB HID v1.11 Keyboard [Corsair CORSAIR M65 RGB ULTRA Gaming Mouse] on usb-0000:06:00.3-4/input3

juanmah, 2023-05-24:
```
> usbhid-dump -m 1b1c:1b9e

001:007:003:DESCRIPTOR         1684930997.323846
 05 01 09 06 A1 01 85 01 05 08 95 03 75 01 19 01
 29 03 15 00 25 01 91 02 95 05 91 03 05 07 75 01
 19 E0 29 E7 95 08 81 02 19 00 29 67 95 68 81 02
 19 87 29 8E 95 08 81 02 C0 05 0C 09 01 A1 01 85
 02 95 01 75 10 15 00 26 FF 03 19 00 2A FF 03 81
 00 C0

001:007:002:DESCRIPTOR         1684930997.324374
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

001:007:001:DESCRIPTOR         1684930997.324609
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

001:007:000:DESCRIPTOR         1684930997.324910
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 08
 15 00 25 01 95 08 75 01 81 02 75 10 95 02 05 01
 09 30 09 31 16 01 80 26 FF 7F 81 06 75 08 95 01
 05 01 09 38 15 81 25 7F 81 06 06 00 FF 09 F1 75
 08 95 05 15 00 26 FF 00 81 00 C0 C0
```

DeN-AlB, 2023-09-07:
Would be great if this device could be supported. :)

TheBuzzSaw, 2023-11-22:
I just bought this device and would love to see it supported. I would love to contribute if I can. I'm software guy but not much of a hardware guy.

rylos, 2024-01-26:
Any update on this support? I'm a developer, is there something I can do?

rylos, 2024-01-31:
Ok here are some wireshark captures of this device under windows i've taken, description is in filenames
[m65_rgb_ultra_wireshark_captures.zip](https://github.com/ckb-next/ckb-next/files/14109587/m65_rgb_ultra_wireshark_captures.zip)

gollum23, 2024-04-13:
Some update about this device?

StefSleepy, 2024-11-19:
Will this be supported in the future? Can we contribute?

Adriankor, 2025-01-12:
Same here, would love to see it supported

jbusich, 2025-02-07:
Glad to see this Request added to the device support list.

spmorton, 2025-02-08:
in which piece of code do you see that?

jbusich, 2025-02-08:
under the general information section supported devices.  They updated the list and included all the ones people have requested.

spmorton, 2025-02-08:
this one still shows requested

[image]

StefSleepy, 2025-02-08:
Until it gets supported (if it gets supported), on OSX you can use the official iCue from corsair to bind the buttons. Its a bit bloated but you can remove any extra software that comes with it after the installation. (for example audio stuff for headphones)

Bi0s-Lain, 2025-04-15:
How can we be mentioned when it will be supported, and what can we do in order to help ?

attilatdeak, 2025-06-22:
What is needed for this device to get into supported status?

`ckb-next-dev-detect - Sun Jun 22 03:00:28 CEST 2025

```
Kernel: Linux attila-arch-mainrig 6.15.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 19 Jun 2025 14:41:19 +0000 x86_64 GNU/Linux

OS Version:
LSB Version:	n/a
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a

USB Devices:

Bus 001 Device 006: ID 1b1c:1b49 Corsair CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard
Negotiated speed: Full Speed (12Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b49 CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard
  bcdDevice            3.24
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard
  iSerial                 3 1101A02AAF791D045F31377EF5001BC3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 (null)
          wDescriptorLength     192
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
[... 306 lines cut ...]
/dev/input/ckb1/productid
1b49
/dev/input/ckb1/serial
1101A02AAF791D045F31377EF5001BC3
/dev/input/ckb1/model
CORSAIR K70 RGB MK.2 Mechanical Gaming Keyboard
/dev/input/ckb1/notify0
/dev/input/ckb1/cmd

Kernel command line:
initrd=\amd-ucode.img initrd=\initramfs-linux.img root=PARTUUID=7af61510-d45d-bc4e-b78b-84fdb2157f00 rootfstype=ext4 rw quiet amdgpu.ppfeaturemask=0xffffffff iommu=1 amd_iommu=on rd.driver.pre=vfio-pci amd_pstate=active

GUI ldd:

Environment:
XDG_SESSION_DESKTOP=gnome
XDG_SESSION_TYPE=wayland
QT_BEARER_POLL_TIMEOUT=-1
XDG_CURRENT_DESKTOP=GNOME
WAYLAND_DISPLAY=wayland-0
GNOME_SETUP_DISPLAY=:1
DISPLAY=:0
QT_IM_MODULE=ibus
```
`

rylos, 2026-02-11:
You can use my fork with full M65 (PID: 0x1b9e) support. Thanks to @martinpal for M65 support, this fork also includes mouse buttons release fix. Link: https://github.com/rylos/ckb-next or PR: https://github.com/ckb-next/ckb-next/pull/1258

Selene0623, 2026-10-07:
I used to use this mouse a lot but my wire started to short out. For anyone else on this mouse is there any progress on merging the PR?

rylos, 2026-10-08:
@Selene0623 just use my repo and build from here, it has the PR merged: https://github.com/rylos/ckb-next