ckb-next/ckb-next #913: Corsair K55 Pro RGB not applying correctly

WorldTeacher, 2022-11-24:
### Affected Devices

Corsair K55 Pro

### Linux Distribution

Arch

### Installation Method

from Source, changed to k55pro branch

### ckb-next version

ckb-next 0.5.0-23-g13b5ffb

### What did you expect?

Expected behavior:
setting the colors to this:
[image]

should make the keyboard blue, but instead it colors it like this:
[image]

I have also tried setting an animation that 

Is there something I can/have to change locally?

### Did you try anything that may have helped?

rebooting, replugging the keyboard and restarting ckb-next do not help.

### Generated Report

[ckb-next-report-1669283779.gz](https://github.com/ckb-next/ckb-next/files/10083236/ckb-next-report-1669283779.gz)

747Cuber, 2022-12-05:
To add on, similar issue using Ubuntu 22.10 via same branch, after messing around, it appears that the zone mapping is inaccurate (one of the zones affects the Lock/M light)
[image]
[image]

747Cuber, 2022-12-05:
Saw that k55pro was merged with master, still having the identical issue

Yueh72, 2022-12-19:
I got it too. After several tests, it seems the different colrs (R,G,B) are merged between different zones.
For exemple, if you change color in zone 1 in ckb, it will change color in zones 1,4 and 5 on the keyboard. Red in zone 1 in ckb will apply red in zone 5 on keyboard and not in zone 1.
It seems to be the same thing with every zones on ckb. Different colors applies to different zones.

scampbellnet, 2023-01-14:
This may be helpful, without actually knowing much about the internals, I played around in the zones and mapped several bytes to the intensity of specific rgb colors for specific portions of the keyboard. 

On my Corsair K55 Pro (1b1c:1ba4) the FF bytes are intensity 0-255 of a component (0 off, 1 almost invisible on up to 255 full brightness).  I didn't do anything with animations, just moved through the bytes of the zones testing their values.  I also spot checked a number of values and didn't see any anomaly. 

Zone 1
        FF---- no obvious effect
        --FF-- number pad red
        ----FF cursor keys green

Zone 2
        FF---- left keyboard red
        --FF-- no obvious effect
        ----FF number pad green

Zone 3
        FF---- mid keyboard red
        --FF-- left keyboard green
        ----FF "1" indicator brightness

Zone 4
        FF---- right keyboard red
        --FF-- mid keyboard green
        ----ff left keyboard blue

Zone 5
        FF---- Cursor keys red
        --FF-- right keyboard green
        ----FF mid keyboard blue

LockBlock-dev, 2023-02-18:
I don't own this keyboard but looking at the code the problem could be related to those lines: https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/daemon/led_keyboard.c#L133-L138
The daemon is assuming all K55 variants are the same: https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/daemon/usb.h#L48
And loops only 3 times (3 lighting zones): https://github.com/ckb-next/ckb-next/blob/c7d04173cc8eb7fa1b6f94ff1a4496a2590be0bc/src/daemon/led_keyboard.c#L85-L94

tatokis, 2023-03-29:
Should be fixed by https://github.com/ckb-next/ckb-next/commit/75f107f228354e1992838083ccf7fa37b9a85204
Someone please also try, with the GUI closed (and assuming the device is ckb1):

`echo -e "active\nrgb b:ffffff" > /dev/input/ckb1/cmd`

and let me know if anything lights up white. There's a 6th lighting zone but I have no idea what it controls (if anything).

tatokis, 2023-03-29:
To clarify, for anyone using -git packages, you should receive this update either after rebuilding (AUR) or next Monday (PPA).

I also dug further, and it appears that the 6th zone is in fact the winlock indicator, which is actually RGB. Can someone please play with the colour value above (ffffff) and verify that it does in fact control the winlock LED?

WickedSmoke, 2023-03-29:
> Someone please also try, with the GUI closed (and assuming the device is ckb1):
> 
> `echo -e "active\nrgb b:ffffff" > /dev/input/ckb1/cmd`

Those 6 bytes set the RGB values of the keypad zone.  They don't change the Windows Lock indicator.

nikokio, 2023-03-31:
On my side, I also have an issue slightly different : I built the project from branch master using the latest commit [9a1b4c](https://github.com/ckb-next/ckb-next/commit/9a1b4ced42d55d5ef1693bed8f65ff18be7bddf1). All is working well except the backlight of the numeric pad (zone 5) : it is always off (no light). The 4 other zones are working well

If I exit ckb-next, lights are working as expected (keyboard firmware driven)

closed 2023-04-01 by tatokis

WickedSmoke, 2023-04-01:
Commit 1a855b1b fixes the main keyboard Lighting zones, but the Windows Lock key is broken.

The Lighting zone color does not change the LED intensity and pressing the key does not toggle the LED.  It generates a Media Volume down code, even though the Binding tab says Windows Lock -> Toggle Windows lock.

WickedSmoke, 2023-04-01:
Also in 1a855b1 the "Insert" label font on the Binding keyboard image is too large.
[image]

hlechner, 2023-04-01:
@WickedSmoke  you probably face the same issue as me in #868 

Try to change this line: 

https://github.com/ckb-next/ckb-next/blob/351f746ec34bdb3b41551e3d8034be28825f44b2/src/daemon/keymap_patch.c#L58

to
```c
{ 114,  "lock",    0, KEY_CORSAIR },
```

WickedSmoke, 2023-04-01:
@hlechner Thanks, 114 does gets the lock toggle working.

After installing the new daemon the large "Insert" key label is also magically fixed.

The Lighting Windows Lock color, however, still does not work.

After a rebuild the large "Insert" key label is also magically fixed.

commit 351f746e (2023-04-01): Apply offset to RGB zones on K55 Pro

Closes #913