ckb-next/ckb-next #915: No devices connected, but I have a K70 RGB Pro

adocampo, 2022-12-25:
### Affected Devices

K70 RGB Pro

### Linux Distribution

Arch

### Installation Method

AUR

### ckb-next version

ckb-next-0.5.0

### What did you expect?

The GUI should show my keyboard and let me configure it.

### Did you try anything that may have helped?

I've rebooted the computer to see if that helped, but no.

### Additional Details

Add any additional details here.

### Generated Report

Drag your generated report here.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/10300624/ckb-next-dev-detect-report.gz)

mnbnd, 2022-12-25:
I have the same issue. It doesn't get detected in OpenRGB both on Windows or Linux if that helps as well.

[ckb-next-report-1672011479.gz](https://github.com/ckb-next/ckb-next/files/10300778/ckb-next-report-1672011479.gz)

Theymoon, 2023-01-22:
I reported the same issue half a year ago (https://github.com/ckb-next/ckb-next/issues/865) but it still doesn't work.

I see that @gravityfargo added K70 RGB Pro to the to-do list some time ago (https://github.com/ckb-next/ckb-next/issues/896), but maybe didn't find the time yet.

Here is an updated report as I updated the system (Debian Sid) several times since my first report.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/10474241/ckb-next-dev-detect-report.gz)

gravityfargo, 2024-05-01:
Duplicate of #865

PR #1030 (2025-03-17): add support for k70 pro
Adds support for the K70 RGB Pro (1bb3)

~All the LEDs are working except the logo light. If there are ideas about how to get the logo light working I'm happy to make the change and test~ Fixed with @JohnoKing's patch.

closes #865
closes #915 
closes #928 
closes #998
closed 2025-03-17 by tatokis