ckb-next/ckb-next #917: Steam Deck Installation

deusnightfire, 2022-12-28:
### Describe your situation in detail.

currently have an issue where I cannot seem to install and stuck at ./quickinstall

it claims that Iconv could not be found. I did a bit more research, noticed that I didn't have the php folder. Couldn't figure out how to install the php folder. Completely stuck right here.

### Additional Details

Add any additional details here.

### Generated Report (if applicable)

Drag your generated report here if it's applicable.

tatokis, 2022-12-28:
https://old.reddit.com/r/SteamDeck/comments/t92ozw/for_compiling_c_code/

Also, I suggest building a package like you would on Arch with makepkg from the AUR instead of using quickinstall.

closed 2022-12-28 by tatokis

deusnightfire, 2022-12-30:
I'm a complete noob. Could you send me reference information on how to build a package Arch with makepkg from AUR? I'm like a complete idiot on this. If not, I'll continue trying to do some research online

grzala, 2023-01-29:
@deusnightfire 
I managed to get the program running on Steam Deck. If I understand correctly, Valve has removed some common libraries that were unnecessary to get the Deck running in order to minimize the size of the OS, so some packages need to be installed or reinstalled.

1. Make sure you run `sudo pacman -S base-devel cmake libsystemd qt5-base zlib libdbusmenu-qt5 quazip libpulse qt5-tools libxcb xcb-util-wm qt5-x11extras git`. When prompted to choose a base-devel package, simply press enter to install all of them

2. Install missing packages / reinstall existing packages
`sudo pacman -S libiconv systemd xorgproto glibc linux-api-headers libgl`

3. Install ckb-next using AUR
`
git clone https://aur.archlinux.org/ckb-next-git.git
cd ckb-next-git
makepkg -si
`

Unfortunately I was not able to use it with my devices, as I found out that my keyboard (K63 wireless) is in experimental stage, and my mouse (Sabre PRO) is not supported :facepalm: