ckb-next/ckb-next #920: Support for pairing

JoePfeiffer, 2023-01-10:
### Enhancement Details

My Dark Core RGB Pro SE has lost its pairing with its dongle, and there does not appear to be a way to reestablish it without booting into Windows and using their iCue software.  Being able to bring it back through ckb-next would be very much appreciated (or if I simply missed how to do it, instructions would also be appreciated!).

### Additional Details

Add any additional details here.

S-Traut, 2023-01-24:
After fiixng a re-pairing issue on my Corsair mouse with ckb-next, I might take a look at implementing a solution for pairing devices. However i'm not sure how this can be handle in a "generic" way, because my pairing method may not work with other devices than the IRONCLAW WIRELESS RGB.

JoePfeiffer, 2023-01-24:
Are you saying you were able to re-pair through ckb-next?  If so, how?  I'm happy to test anything you come up with using my mouse.

S-Traut, 2023-01-24:
Yes by editing some code and using bragi functions, the goal is to rewrite the pairid and the encryption key of both devices. I've done it manually just for fixing my own mouse right now, but I might take a look at implementing something that handle most of the bragi protocol devices.

S-Traut, 2023-04-10:
@JoePfeiffer FYI, if you really want to pair through ckb-next, you can grab my branch and build it yourself
#952

closed 2024-05-02 by gravityfargo

RokeJulianLockhart, 2024-05-02:
> https://github.com/ckb-next/ckb-next/issues/920#event-12676805328

@gravityfargo, what commit or PR completed this?

gravityfargo, 2024-05-02:
[image]