ckb-next/ckb-next #921: add "CORSAIR DARK CORE RGB PRO" mouse support

philippe9999, 2023-01-18:
### What kind of device is it?

Mouse

### What is the name of the device?

CORSAIR DARK CORE RGB PRO

### Link to Corsair's Product Page

https://www.corsair.com/ca/en/Categories/Products/Gaming-Mice/Wireless-Gaming-Mice/DARK-CORE-RGB-PRO-Wireless-Gaming-Mouse/p/CH-9315411-NA#tab-overview

### Additional Details

This mouse has id 1b80 connected to usb port with wire and 1b81 with the dongle wireless. 

Fell free to ask question or more info.

Thanks

### Generated Report

Drag your generated report here.
[ckb-next-report-1674011561.gz](https://github.com/ckb-next/ckb-next/files/10441894/ckb-next-report-1674011561.gz)

JoePfeiffer, 2023-01-24:
Is this addressed by PR #855?  I've been using this mouse with ckb-next (following the instructions in that PR) for some time now.

philippe9999, 2023-01-24:
Bonjour

Thanks for the reply. Experimental mode is activate by systemd and systemctl status confirm the activation. I use OpenSuse Tumbleweed up to date with ckb-next version 0.5.0-1.2. My keyboard K55 work well. 

As per supported device list, my mouse ( ID 1b1c:1b80 Corsair CORSAIR DARK CORE RGB PRO Gaming Mouse ) is not in the list. Maybe my mouse get a different id from yours.

Can you confirm the id of your mouse.

Thanks

JoePfeiffer, 2023-01-24:
Ah, that does appear to be a difference.  My wired connection is 
1b1c:1b7e Corsair CORSAIR DARK CORE RGB PRO SE Gaming Mouse and my dongle is
1b1c:1b7f Corsair CORSAIR DARK CORE RGB PRO SE Gaming Dongle
No idea why they'd be different, since the model name appears to be the same.

philippe9999, 2023-01-24:
Not exactly the same, mine is a PRO only model not SE one.  ( PN CH-9315411 ).

Yueh72, 2023-02-01:
This project seems to be abandonned. Nobody touch it since october 22

Ravenslofty, 2023-02-01:
It's not abandoned. Please don't say that it is; the developers are humans too and they have priorities and real life to attend to. It's a free software project developed in our spare time. Sometimes we just don't have very much spare time.

Clofriwen, 2023-02-04:
Hello,

My PR #855 support this device SE and non SE version.
@colbiedison have a non SE version and it's look working.

In source code, you have :
`P_DARK_CORE_RGB_PRO :  0x1b80 & 0x1b81` for wired and wireless mode of non SE
`P_DARK_CORE_RGB_PRO_SE : 0x1b7e & 0x1b7f` for wired and wireless mode of SE

I invite you to test my PR.

philippe9999, 2023-02-04:
Salut

Test your PR with success. Mouse is well recognize and all function seem to work. I will carry more test next week to check all options.

Thanks for your work and the time to do that PR.

philippe9999, 2023-02-04:
This issue is closed , the mouse is supported by PR #855.

closed 2023-02-04 by philippe9999