ckb-next/ckb-next #923: Ez way to script or display polled values every 0.x seconds.

NicTanghe, 2023-01-24:
### Enhancement Details

Hello on windows i have this layout setup where i display my CPU temp / usage on my F1-F11 and F5-F8 kays respectively.

As well as a binary clock and other usefull metrics like screencapping my life bar in and displaying how much health I have on my keyboard.

Not having these makes the rgb lighting seem kinda useless.

It whould be great if some sort of UI is added to ad this functionality or if someone can point me to where the current animations are so i can try and make one myself.

### Additional Details

Add any additional details here.

NicTanghe, 2023-03-02:
Annyone ?

tatokis, 2023-03-03:
https://github.com/ckb-next/ckb-next/wiki/Animations#pipe